OpenStickCommunity/GP2040-CE #288: Abnormal Turbo LED behaviour with PWM Player LEDs

BolSadguy, 2023-05-27:
_LED Positive to 3.3v, LED Negative to GPIO:_
Nothing set to turbo: LED off **(expected)**
Button set to turbo but not being pressed: LED off **(unexpected, should be on with no blinking)**
Button set to turbo and is pressed: LED blinks on and off (dim) **(blinking expected, dimness unexpected)**

_LED Positive to GPIO Pin, LED Negative to GND:_
Nothing set to turbo: LED on (bright) **(unexpected, should be off)**
Button set to turbo but not being pressed: LED on (bright) **(expected)**
Button set to turbo and is pressed: LED flashes on and off (bright) **(expected)**

Pins used:
Player LED 1 through 4 = GPIO 26 through 29
Turbo = GPIO 23
Turbo LED = GPIO 24

TheTrainGoes, 2023-05-29:
Through some chatting this only appears to be an issue when all 5 LEDs are connected (PLED 1-4 + Turbo).

Will test on my own unit and see if I can replicate when I have some time

TheTrainGoes, 2023-07-21:
We have tested this on the new v0.7.2 RC4 and cannot replicate the issue.  I am going to close for now and if you can still replicate please open a new issue and we can check it out in more depth.

closed 2023-07-21 by TheTrainGoes