OpenStickCommunity/GP2040-CE #293: Web UI Revamp 2023

Fortinbra, 2023-05-28:
Placeholder for the revamp of the WebConfig UI that's currently in progress.

FeralAI, 2023-05-29:
Here's a very rough prototype of the new Board Config page:

[image]

fatihG, 2023-05-29:
I worked on a mockup the other day when I asked about the UI, prior to getting a reply. 

Ended up not continuing with what I had in mind, but ill post what I had at the time, for your consideration. 

[image]

The idea being that the layout created in this editor will be used throughout every other page that references the layout. 
Everything would be tied to a "position/switch" which are ordered from old to new. 
To these positions/switches, you set the pin and function. 

This way, the end user has a visual representation of their set up and will make making layouts easier, without needing to cross reference. 

Another thing is that pages can be combined together. 
Extra button add on? Simply drop another button in there. 
Turbo? Just drop in another button. 
Player indicator? Drop down somewhere on the layout. 

The LED layout editor could simply be a separate tab on the same page. Hell, LED options could be on the buttons themselves directly as well. 
For global changes those could be done by a small menu on the button left of the window. Ie. for a gradient.  
Since the end user will be creating their layout visually as well, the gradient will be more accurate as it could pull the transformation data from the graphics and use that to calculate what colour which button needs to be according to their absolute positions. 

I realize my suggestion is a lot of work, less 'data' orientated and it is basically asking for an editor that only a small percentage of a niche portion of a specific game genre would ever put their eyes on. But here it is regardless. 

-- 

To what you have right now, are other pico's being considered? What if someone is using the RP2040-Zero or the other variant that are supported?

FeralAI, 2023-05-29:
Thanks for the input @fatihG!

There is a controller editor being work on for the RGB LED Configuration portion of the redesign. That interface will faciliate a lot of what you're asking for once complete. I hesitate to mention all the features it can be used for because I don't know what will be implemented by the time this update is finished, but it will be a flexible controller design minus the pin mapping in your mockup.

I believe the low-level hardware configuration of pin mapping and expansion port configuration should be kept separated in the new `Board Config`, while other things like LEDs, button mappings, etc. can be handled in a new `Controller Config` based on the RGB LED designer work. This way a user doesn't accidentally remove/change a core feature that needs to be displayed on the Controller Config.

> To what you have right now, are other pico's being considered? What if someone is using the RP2040-Zero or the other variant that are supported?

Every type of board that has a release build for GP2040-CE is being considered for the visual Board Config. There is an SVG spec and a conversion process being defined so board makers can export their boards from their CAD software and convert for use in the web app and have it "just work" with all the highlighting and click events. I'm currently working with Pico, RP2040-Zero and full arcade board SVGs for dev and testing.

SavageCore, 2023-06-23:
For the automated image optimisation, shall we just enable [Imgbot](https://github.com/marketplace/imgbot) on this repo? It did wonders on #357

TheTrainGoes, 2023-09-06:
We are going to close this for now and re-eval the current state of the web-config.

closed 2023-09-06 by TheTrainGoes