OpenStickCommunity/GP2040-CE #294: Full analog support

Fortinbra, 2023-05-28:
Placeholder for completing the analog stick/trigger support.

TheTrainGoes, 2023-06-18:
We want to look into something before doing this - It is possible that dual analog is possible through a broken out embedded RP2040.  

We would like to add an action item to this to expand the current analog offering to take advantage of broken out RP2040 chips to see if things can be handled as is.

TheTrainGoes, 2023-06-30:
We have a working PoC for dual analog support now via a broken out RP2040 chip that has access to all of the ADC pins.

mordae, 2023-07-13:
Hi. I've just accidentally stumbled upon this project and wanted to chime in.

It is possible to use any RP2040 GPIO pin as ΣΔ ADC. To get an idea, check out [this simulation](https://tinyurl.com/2ecqbbp7) I've put together.

As for the implementation, the simplest way would be to:

- Configure the GPIO for output
- Enable pull-down
- Emit 1 (to quickly charge the capacitor)
- Configure GPIO for input (to slowly drain the capacitor)
- Read GPIO until it switches from 1 to 0, incrementing a counter
- Counter value is a usable readout

You can experiment with the passives to get a feel for what would work best.

The ideal implementation would be a little more involved. I have used PIO to get best results. In a nutshell:

- Configure the GPIO for input and use the "bus keep" mode by enabling both pull-up and pull-down. Now when you output something on the GPIO, you are actually controlling the pull-up/down and thus have about ~50kΩ series resistance. This is going to smoothly steer the capacitor voltage.
- Enable PIO for the pin and write a (possibly single instruction) program that outputs negation of the pin back to itself. Since we are in the bus keep mode and there is a relatively large capacitor, we have just constructed a (rather gentle) oscillator.
- Write another PIO program that continuously decrements one of the registers when the pin is high and the other when it is low using branch on pin value. To take measurements:
  - Inject instructions to reset the registers to 0.
  - Let it run for a while, possibly doing other stuff such as checking buttons and so on. Ideally it should run for 10 ms so that you get 100 Hz output, but it's really up to you.
  - Inject push X, push Y to get the counters to the FIFO and then read them from there. Subtract them from 0 to get positive numbers. Then subtract one from the other to get your readout.

You need to calibrate every stick before it gets usable. Depending on capacitor and joystick tolerances there will be differences between readouts of different axes, so you really need to calibrate all of them to get minimum, maximum and center points.

This mechanism is similar to how joysticks were read back in the [Game port](https://en.wikipedia.org/wiki/Game_port) days.

TheTrainGoes, 2023-07-21:
We have now rolled out dual analog for RP2040 based devices that have access to all four ADC pins.

osdx190, 2023-07-25:
**Hi, I've tried your latest update and I've had a problem with the analog ones not giving the maximum values** and it would be because they are 60° resistors on the joystick. I have mapped it with windows and it works but I use it in a game or program it doesn't work at all there are sides that give me like 93% movement, one side gives me 70% and others 81%. these are my external joystick calibration ideas. Although it occurred to me to add capacitors as in the original boards.
[image]
**and i have managed to add an adc on the standard raspberry pi pico board.**

[image]

TheTrainGoes, 2023-09-06:
We are going to close this since we have dual analog now.  Once we have the expansion board ready we can look at doing trigger support.

closed 2023-09-06 by TheTrainGoes