OpenStickCommunity/GP2040-CE #297: LEDs do not work when joystick is set to LS (DP works)

arntsonl, 2023-05-31:
This could be a new bug, switching the joystick to LS is supposedly causing the LEDs to no longer fire for the joystick. Switching back to DP then fixes the issue?

Will need to verify locally.

FeralAI, 2023-05-31:
This is because LEDs respond to the gamepad report instead of raw inputs. The WIP LED refactor is addressing this by mapping LED raw inputs (GPIO, expander, analog, etc.) to the LED nodes they affect.

The current LED presses take into account SOCD, so this new approach may need to factor that in depending on if we want to see LED changes based on physical pressed or logical controller outputs.

TheTrainGoes, 2026-11-05:
This works now as part of the RGB refactor

closed 2026-11-05 by TheTrainGoes