OpenStickCommunity/GP2040-CE #299: please help me Failed building

chiyu521, 2023-06-04:
please help me compile error

Collecting protobuf>=3.19 (from -r /data/home/chiyu/pico/GP2040-CE/lib/nanopb/extra/requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/1d/af/4e351c0157d5a0440dc3ac0b3c1fdaa036b3b446bf98310b5e3d6d9d2733/protobuf-4.23.2-py3-none-any.whl
Collecting grpcio-tools>=1.47.0 (from -r /data/home/chiyu/pico/GP2040-CE/lib/nanopb/extra/requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/ab/57/0a02eb41926c19809e40217f2b8499243db77c683f6f8e24ee6a3d0c7407/grpcio-tools-1.55.0.tar.gz
Collecting grpcio>=1.55.0 (from grpcio-tools>=1.47.0->-r /data/home/chiyu/pico/GP2040-CE/lib/nanopb/extra/requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/76/63/d025fb7599379765d6437e5aab3644955248ea89d2f84f4f8a9762d99c52/grpcio-1.55.0.tar.gz
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from grpcio-tools>=1.47.0->-r /data/home/chiyu/pico/GP2040-CE/lib/nanopb/extra/requirements.txt (line 2)) (40.8.0)
Building wheels for collected packages: grpcio-tools, grpcio
  Running setup.py bdist_wheel for grpcio-tools ... error
  Complete output from command /data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-61bigi0z/grpcio-tools/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-ofm8rkz9 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for grpcio-tools
  Running setup.py clean for grpcio-tools
  Running setup.py bdist_wheel for grpcio ... error
  Complete output from command /data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-61bigi0z/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-5xpjdqs0 --python-tag cp37:
  Found cython-generated files...
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio-tools grpcio
Installing collected packages: protobuf, grpcio, grpcio-tools
  Running setup.py install for grpcio ... error
    Complete output from command /data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-61bigi0z/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-8yev4qa8/install-record.txt --single-version-externally-managed --compile --install-headers /data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/include/site/python3.7/grpcio:
    Cython-generated files are missing...
    We could not find Cython. Setup may take 10-20 minutes.
    running install
    running build
    running build_py
    running build_project_metadata
    creating python_build
    creating python_build/lib.linux-x86_64-3.7
    creating python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_utilities.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_simple_stubs.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_grpcio_metadata.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_server.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_compression.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_interceptor.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_channel.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_runtime_protos.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_typing.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_auth.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_common.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/_plugin_wrapping.py -> python_build/lib.linux-x86_64-3.7/grpc
    copying src/python/grpcio/grpc/__init__.py -> python_build/lib.linux-x86_64-3.7/grpc
    creating python_build/lib.linux-x86_64-3.7/grpc/experimental
    copying src/python/grpcio/grpc/experimental/session_cache.py -> python_build/lib.linux-x86_64-3.7/grpc/experimental
    copying src/python/grpcio/grpc/experimental/gevent.py -> python_build/lib.linux-x86_64-3.7/grpc/experimental
    copying src/python/grpcio/grpc/experimental/__init__.py -> python_build/lib.linux-x86_64-3.7/grpc/experimental
    creating python_build/lib.linux-x86_64-3.7/grpc/beta
    copying src/python/grpcio/grpc/beta/interfaces.py -> python_build/lib.linux-x86_64-3.7/grpc/beta
    copying src/python/grpcio/grpc/beta/_server_adaptations.py -> python_build/lib.linux-x86_64-3.7/grpc/beta
    copying src/python/grpcio/grpc/beta/_metadata.py -> python_build/lib.linux-x86_64-3.7/grpc/beta
    copying src/python/grpcio/grpc/beta/utilities.py -> python_build/lib.linux-x86_64-3.7/grpc/beta
[... 186 lines cut ...]
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/lib/python3.7/site-packages/setuptools/command/build_ext.py", line 78, in run
        _build_ext.run(self)
      File "/usr/lib/python3.7/distutils/command/build_ext.py", line 340, in run
        self.build_extensions()
      File "/tmp/pip-install-61bigi0z/grpcio/src/python/grpcio/commands.py", line 279, in build_extensions
        self.extensions = try_cythonize(self.extensions)
      File "/tmp/pip-install-61bigi0z/grpcio/src/python/grpcio/commands.py", line 197, in try_cythonize
        compiler_directives=cython_compiler_directives)
      File "/tmp/pip-install-61bigi0z/grpcio/.eggs/Cython-3.0.0b3-py3.7.egg/Cython/Build/Dependencies.py", line 1134, in cythonize
        cythonize_one(*args)
      File "/tmp/pip-install-61bigi0z/grpcio/.eggs/Cython-3.0.0b3-py3.7.egg/Cython/Build/Dependencies.py", line 1301, in cythonize_one
        raise CompileError(None, pyx_file)
    Cython.Compiler.Errors.CompileError: src/python/grpcio/grpc/_cython/cygrpc.pyx
    
    ----------------------------------------
Command "/data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-61bigi0z/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-8yev4qa8/install-record.txt --single-version-externally-managed --compile --install-headers /data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/include/site/python3.7/grpcio" failed with error code 1 in /tmp/pip-install-61bigi0z/grpcio/

mthiesen, 2023-06-04:
What operating system are you using? Which version?

chiyu521, 2023-06-06:
deepin similar to ubuntu

bsstephan, 2023-06-06:
You might need to `sudo apt-get install python3-wheel`.

mthiesen, 2023-06-06:
Also, could you try updating your Python version? 3.7 is nearly 5 years old and is entering End Of Life later this month.

chiyu521, 2023-06-07:
sudo apt-get install python3-wheel
did not solve the problem

mthiesen, 2023-06-13:
On my Ubuntu machine I am using Python 3.8. The GitHub Actions build nodes use Python 3.10. Both should work fine.

chiyu521, 2023-06-14:
I have upgraded Python 3.11.4 output result is like this

[ 58%] Setting up Python Virtual Environment
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/data/home/chiyu/pico/GP2040-CE/build/lib/TinyUSB_Gamepad/venv/bin/python3.7', '-Im', 'ensurepip', '--upgrade', '--default-pip']

make[2]: *** [lib/TinyUSB_Gamepad/CMakeFiles/TinyUSB_Gamepad.dir/build.make:85：lib/TinyUSB_Gamepad/venv/environment.txt] error 1
make[1]: *** [CMakeFiles/Makefile2:2354：lib/TinyUSB_Gamepad/CMakeFiles/TinyUSB_Gamepad.dir/all] error 2

chiyu521, 2023-06-14:
python3-venv 3.7.3-1

mthiesen, 2023-06-14:
Did you run `sudo apt-get install python3-venv` as suggested by the error message?

Also, it might be necessary to delete the `build` directory and create a new one after the install of `python3-venv`.

chiyu521, 2023-06-14:
SUDO APT Install Python3-Venv
Python3-Venv is the latest version (3.7.3-1).

I don't know if there is a way to install version 3.11.4

chiyu521, 2023-06-15:
Newly installed UBUNTU 22.04.2 and rebuilt the environment, the problem was solved

closed 2023-06-15 by chiyu521

mthiesen, 2023-06-16:
Glad you got this working!