OpenStickCommunity/GP2040-CE #301: 4-way input cleaning - new SOCD mode for 4-way games

zzymyn, 2023-06-05:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

For better support for arcade games that only support 4-way stick input, at the moment if you hold left and tap down, the game will get (left, down/left, left).

This is an issue for arcade games such as Tetris The Grand Master as they were designed to be played with a joystick using a 4-way gate and when it sees (left, down/left, left), it ignores the down input and does nothing.

### Describe the solution you would like to see

I would like a new SOCD mode that removed diagonal inputs and only sends the latest pressed cardinal direction to the game. So for the example above if you hold left and tap down, the game should get left, down, left.

The way I think it should work would be: for the raw inputs of left, right, up and down, whichever direction was pressed soonest is the only direction sent to the game.

### Can you submit a pull request?

Eventually

### Are you planning on working on this feature yourself?

Yes

TheTrainGoes, 2023-06-05:
Hi @zzymyn ,

This is a request we have seen a few times.  Would be great to have it added in.  

Possibly could be setup as an add-on or joystick mode as well.  Not sure a custom SOCD for it would be the easiest route.

bsstephan, 2023-06-05:
Yeah, I think SOCD is enticing because it's straightforward and `runSOCDCleaner` would be a good spot for it, but I agree with Train in that a 4-way "joystick mode" might be better because then you'd get the behavior regardless of selected SOCD mode (and technically the 4-way problem isn't SOCD anyway). I think you'd just have to do another check early on in `Gamepad::process()`. The only wrinkle is I don't remember where the prior state is retained.

Either way, good request! I can think of games I play semi-regularly that'd benefit from this.

zzymyn, 2023-06-05:
I threw together something quick just to try it out in-game: https://github.com/OpenStickCommunity/GP2040-CE/compare/main...zzymyn:GP2040-CE:fourway (I haven't looked at the source code for this before now so no idea if this is the best way of doing it).

Had a quick game of TGM2 and it feels so much nicer than the 8-way input.

bsstephan, 2023-06-08:
@zzymyn I looked at the code, I like the idea but want to take a crack at a clearer implementation. I'm going to assign this to myself and see if I come up with a method that works the same as yours in a bit less code.

bsstephan, 2023-06-09:
@zzymyn let me know if https://github.com/OpenStickCommunity/GP2040-CE/pull/320 works just as well for you. I implemented it as a separate option outside of the SOCD settings.

zzymyn, 2023-06-11:
Hey I already sealed up my joystick so can't flash a new firmware on it at the moment. Will need someone else to test it.

TheTrainGoes, 2023-06-11:
@zzymyn you should be able to hold Start + select + up while plugging in to put in BootSel mode

zzymyn, 2023-06-11:
Ah thanks didn't realise. I can try tomorrow hopefully. @bsstephan will I be able to set up a hot key to swap between 4-way mode and regular mode?

bsstephan, 2023-06-12:
@zzymyn - not *yet*, but I am also have hotkey improvements in the works, and I would seek to make it possible once those are merged.

bsstephan, 2023-06-13:
I did implement a hotkey action for toggling 4-Way mode, and then discovered a bug in main, so my PR will eventually have hotkey actions for both normal and DDI 4-Way, but not until the bugfix goes in, because all the conflicts are getting to be a bit much for me to keep track of.

closed 2023-07-31 by TheTrainGoes