OpenStickCommunity/GP2040-CE #308: Phantom inputs

wrennnnnn, 2023-06-07:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.2RC1
* Name of device: Integrated Pico Fighting Board
* Link to where you purchased the device: https://jasenscustomsarchive.com/product/integrated-pico-fighting-board/
* Is this a custom build?: Yes, created by me.
* Operating System: Windows
* Browser: Firefox

### Expected Behavior

no inputs should be held. player leds and turbo should be working.

### Current Behavior
Phantom, random inputs (left) on Dpad while in xinput, ps4, or ps3 mode. All player LEDs are lit up.

### Steps to Reproduce

Boot into PS3, PS4, Xinput mode with attached UF2 or boardconfig. 

### Screenshots & Files

https://github.com/OpenStickCommunity/GP2040-CE/assets/67722465/a1b5096e-7ac4-498b-a5c0-d16391109e42
https://cdn.discordapp.com/attachments/1049370339060240474/1116043180232687787/GP2040-CE_0.7.2_PicoFightingBoard.uf2
https://cdn.discordapp.com/attachments/1049370339060240474/1116043180496932864/gp2040ce_backup_20230607162642037.gp2040

arntsonl, 2023-06-07:
The cause is the Turbo Add-on.

For some reason, if Turbo Add-on pin is set and Turbo is enabled, we get phantom left inputs.

I have a feeling the left input might be changing the gamepad dpad state? There's one instance where it does that, but that shouldn't be happening here.

TheTrainGoes, 2023-06-08:
This has been addressed in v0.7.2 RC2 - https://github.com/OpenStickCommunity/GP2040-CE/releases

closed 2023-06-08 by TheTrainGoes