OpenStickCommunity/GP2040-CE #309: Hotkeys settings are not saved in the backup file

LetramChi, 2023-06-07:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.1 (0.7.2 RC1)
Name of device: Raspberry Pi Pico
Is this a custom build?: Yes
Operating System: Windows 10
Browser: Chrome

### Expected Behavior

When uploading the backup file, you should also upload the hotkey settings.

### Current Behavior

Loads all button settings, display, addons, etc... except the hotkeys

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Step 1: Generate backup file with hoykey settings ready.
2. Step 2: Load backup file
3. Step3: Check in the Settings section (Hotkey Settings)
4. All options are with the information "No Action"

### Screenshots & Files

[image]
[image]

bsstephan, 2023-06-08:
I think we can fix this in 0.7.3, having discussions around how this page could be unified with protobuf changes.

bsstephan, 2023-12-27:
Just pinging here that this is still unfixed as we have not yet done #657, but it is on my list of things to do Soon (TM).

TheTrainGoes, 2024-02-04:
This issue has been resolve and hotkeys are now part of the backup.

closed 2024-02-04 by TheTrainGoes