OpenStickCommunity/GP2040-CE #314: Hotkeys don't change when Touchpad and Share are switched in PS4 Mode

jazonandes, 2023-06-08:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.2-RC2
* Name of device: Custom
* Link to where you purchased the device: Custom
* Is this a custom build?: Yes
* Operating System: Windows 10
* Browser: Firefox

### Expected Behavior

When Touchpad and Share are switched, the Web Configurator states that A2+S2 / Touchpad+Options should be held to access the first set of hotkeys.

### Current Behavior

When Touchpad and Share are switched, S1+S2 / Share+Options must still be held in order to access the hotkeys.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Change input mode to PS4.
2. Switch Touchpad and Share in the Web Configurator.
3. Attempt to access hotkeys by holding A2+S2 / Touchpad+Options.

### Screenshots & Files

[image]
[image]
[image]

bsstephan, 2023-06-08:
Sorry, I think this is an unfortunate side effect of the swap feature. The option swaps the *function* of the buttons, but the button themselves are the same, and the hotkeys react to the physical button pressed, not the result.

jazonandes, 2023-06-08:
Understandable. Wish I was smart enough to edit the Web Configurator so that the labels under Hotkey Settings didn't change when Touchpad and Share are switched, but that's beyond me. Thank you for the help!

teikai1216, 2023-06-08:
#317 will fix the hotkey when you swap the touchpad and share buttons in PS4 mode.

bsstephan, 2023-06-08:
I left a comment on the PR, I think it's better to wait for a different feature rather than changing the button mapping contributions directly. I can't grok right now if the UI changes still make sense in light of that.

TheTrainGoes, 2023-12-10:
This has been addressed and merged into main

closed 2023-12-10 by TheTrainGoes