OpenStickCommunity/GP2040-CE #327: Can add xinput mode rumble support？

mlpzaq9998, 2023-06-13:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Describe alternatives you've considered

[Let us know about other solutions you've tried or researched.](https://github.com/dmadison/ArduinoXInput)

### Can you submit a pull request?

Yes

### Are you planning on working on this feature yourself?

No.

mlpzaq9998, 2023-06-13:
https://github.com/dmadison/ArduinoXInput
https://github.com/dmadison/ArduinoXInput_Teensy

mlpzaq9998, 2023-06-13:
Is it possible to add a temperature sensor to automatically adjust the gamepad cooling?1-Wire or i2c interface。

> Short answer, No.
> 
> Longer answer: We also have to be cognizant of a lot of different variables. Using the drv chip as the example. It's is $2.90(USD) on DigiKey for individual chips. Adding an I²C device does not use any additional pins from the pico/rp2040, and it already has an Arduino library that fully supports it, so it's only a matter of porting that library over to our uses. All of this is before even considering that we also need to add the ability to read the haptic data coming from the host, and send it in the correct form to the driver so it can drive the motor. Everything is iterative, and nothing is set in stone, so the drv2605L makes for a good first step, and we don't know where their staircase leads. So for now, the simplest and fastest route to a usable feature comes first, then iterating on that feature to make it better will come later.

Is it possible to add a temperature sensor to automatically adjust the gamepad cooling?1-Wire or i2c interface。
1-Wire interface, there is a very small DFN1608-2 package NST1001-QDNR.
https://github.com/JoeyStrandnes/NST1001_Arduino-Driver

Fortinbra, 2023-06-13:
> Is it possible to add a temperature sensor to automatically adjust the gamepad cooling?1-Wire or i2c interface。
>
>
>
> Is it possible to add a temperature sensor to automatically adjust the gamepad [...]

That's a completely separate topic and request. Please try to keep these issues centered on one thing

TheTrainGoes, 2023-09-06:
We are going to close this issue as not planned currently.  

Once we have the expander board we will look at the extended features of a controller we can bring in with it.

closed 2023-09-06 by TheTrainGoes