OpenStickCommunity/GP2040-CE #328: Vita TV / PlayStation TV support

Soundtoxin, 2023-06-13:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes.

(FYI, your site appears very broken right now, the main page especially, had to check it via archive.org, but some specific pages like the "Usage" page still load up)

### Which problem is this feature request solving?

When connecting my Flatbox Rev 5 to a Vita TV, I am unable to get it to work as a controller.

### Describe the solution you would like to see

If there were a way for the firmware to spoof a DualShock 3 or DualShock 4, I believe the controller would work.

### Describe alternatives you've considered

It may also be possible for someone to write a homebrew plugin for the Vita (TV) to support the controller as it is. I have seen support for [8bitdo](https://github.com/Ibrahim778/8BitVita) and [Xbox](https://github.com/Ibrahim778/X1Vita) controllers added this way while researching controller problems (although they appear to be for bluetooth and not wired connections).

### Can you submit a pull request?

No.

### Are you planning on working on this feature yourself?

No.

Fortinbra, 2023-06-13:
We already have a PS4 support as a DS4 controller. 

For now, any and all Playstation related changes that are not bugs related to our existing features, will be pushed back until after our 1.0 release .

Soundtoxin, 2023-06-13:
>can you please let us know what URL you were going to?

https://gp2040-ce.info/#/ is the URL that gave me a problem. For some reason it's having issues on my computer using qutebrowser (shows a 404 error), but seems to load on my phone. I noticed another issue just on my PC where the button labels for switching input modes were invisible, but they showed on my phone. Very odd. I'm using Fennec F-Droid (Firefox fork) on my phone. qutebrowser should be pretty similar to chromium on the backend, I believe.

>DINPUT mode (PS3) should work just fine for the Vita TV.

I tried both this and the PS4 mode and my inputs don't seem to do anything. If I go to xinput mode I get to see a message that the device isn't supported in the corner, but the other modes don't give me any feedback. 
I've verified on my computer that the mode is switching properly by comparing dmesg output and seeing the controller shows up differently. The cable and port I'm using with the Vita TV are known to be working with another controller (standard DualShock 4).

Has GP2040-CE been tested with a Vita TV specifically? I think it's more strict about supported controllers (third party mostly) than a PS3.

TheTrainGoes, 2023-06-13:
No it has not been tested directly on a Vita TV.  I have not owned one of those for many years.  

Is yours modified at all?  There may be a way to enhance controller compatibility if so.  

If you have checked all of the input modes and they are not working then I would say GP2040-CE is not compatible with the Vita TV.  

You could try using the original GP2040 v0.4.3 from the original repo to see if it works though.  We made some changes along the way to how the PS3 controller works.  

Also, are you pressing the home button after it is plugged into the Vita TV in PS3 mode?

Pelsin, 2023-06-13:
Are you using **safari** when visiting: https://gp2040-ce.info/#/ @Soundtoxin?
I think there might be some issues when going to root url

bsstephan, 2023-06-13:
I have a Vita TV (still plugged in, even!) and will give it a try if I get a spare moment.

Soundtoxin, 2023-06-13:
> Are you using **safari** when visiting: https://gp2040-ce.info/#/ @Soundtoxin? I think there might be some issues when going to root url

@Pelsin  No, the browser I am using is called "[qutebrowser](https://qutebrowser.org/)". I believe the site worked fine for me in the past in this browser, but I'm not sure how long ago that was.

Pelsin, 2023-06-13:
Seems to be the same issue with safari on my phone and desktop, i'll create a separate issue for it!

bsstephan, 2023-06-13:
Confirmed, DInput/PS3 and PS4 mode seemingly do nothing on my Vita TV, XInput pops up an unsupported message. 0.7.2.

Soundtoxin, 2023-06-14:
For now, I was able to get my Flatbox working in xinput mode after setting up the "vixen" plugin on my Vita TV. 

https://github.com/isage/vixen

This seems like a decent workaround for now, but getting the DInput/PS3 mode working in the future seems worth it as well, so I'd like to keep the issue open if that's okay. I understand it's low priority for now while other things are worked on.

TheTrainGoes, 2023-06-14:
@bsstephan totally ok with me.  

If you have a chance try out the original v0.4.3 to see if that works.  We made some changes to PS3 to get the home button working.

TheTrainGoes, 2023-09-05:
I'm going to close this issue for now since there is a working solution in the comments.

closed 2023-09-05 by TheTrainGoes