OpenStickCommunity/GP2040-CE #337: Assignable multimedia keyboard controls

totozero, 2023-06-14:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes. Only one extra key is allowed, as of now.

### Which problem is this feature request solving?

Control multimedia keyboard functions on button press.

### Describe the solution you would like to see

I'd like to be able to map multimedia keys to button functions, especially volume up and volume down while in keyboard mode.
It would be useful in an arcade cabinet running a pc, as there is no interaction with a keyboard in such environment, only sticks an buttons.

### Describe alternatives you've considered

- Separate board for just multimedia controls (as an Arduino Leonardo with hoodloader https://github.com/NicoHood/HID)
- ahk script to intercept gp2040 keypress and remap it into multimedia function // or same solution with xpadder

### Can you submit a pull request?

No.

### Are you planning on working on this feature yourself?

No.

TheTrainGoes, 2023-06-15:
Just to narrow this down a bit, for the time being this would need to be a HID-Keyboard mode only thing.

I am going to put the base bounty as "add vol up and vol down".

totozero, 2023-06-15:
Yes, I second this, only for keyboard mode. It would be a bad idea to mix inputs from different modes.

TheTrainGoes, 2023-09-05:
This was completed and merged into main

closed 2023-09-05 by TheTrainGoes