OpenStickCommunity/GP2040-CE #342: HOTKEY_CAPTURE_BUTTON doesn't do anything

bsstephan, 2023-06-16:
I noticed this while doing all the other hotkey changes I've been working on, but didn't want to muddle PRs. `case HOTKEY_CAPTURE_BUTTON` doesn't do anything other than `break`.

I'm guessing it should `state.buttons |= options.switchTpShareForDs4 ? GAMEPAD_MASK_A2 : GAMEPAD_MASK_S1`, to reflect how we made the button consistent and pushed the TP/Share toggle into the PS4 report.

bsstephan, 2023-06-21:
This issue is blocked by #360 as I broke the feasibility of the intended fix.

bsstephan, 2023-07-07:
Addressed in https://github.com/OpenStickCommunity/GP2040-CE/pull/394 --- just waiting for a release to close this.

TheTrainGoes, 2023-07-21:
This has been rolled into RC, closing issue

closed 2023-07-21 by TheTrainGoes