OpenStickCommunity/GP2040-CE #345: Joystick Selection Slider Not Working

mdiph, 2023-06-18:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

* Firmware Version:  0.7.2
* Name of device: Waveshare RP2040-Zero
* Link to where you purchased the device: Local Store
* Is this a custom build?: No
* Operating System: Windows 10
* Browser: Chrome

### Expected Behavior

The joystick should function when the button for pin 26 is pressed.

### Current Behavior

When the button that is connected to the Pin 26 is pressed, it doesn't do anything, but when i disable the joystick selection slider, and change it to the original setting Pin 26 (L3), the L3 is working fine. tested it on the older firmware 0.6.2, the joystick selection slider is working fine, and when pressed it changed the direction (UP, DOWN, etc) into a joystick.

TheTrainGoes, 2023-06-18:
Hi @mdiph ,

Thank you for the report - we will take a look into this when we have some time.

TheTrainGoes, 2023-07-21:
I have verified that this is not working as intended and we will look into it.

TheTrainGoes, 2023-07-25:
This has been fixed!  Resolving.

closed 2023-07-25 by TheTrainGoes