OpenStickCommunity/GP2040-CE #346: Controller doesn't remember to switch Touchpad/Share when changing to PS4 Mode upon plugging in

jazonandes, 2023-06-18:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.2
* Name of device: Custom
* Link to where you purchased the device: Custom
* Is this a custom build?: Yes
* Operating System: Windows 10
* Browser: Firefox

### Expected Behavior

When holding B4 upon plugging in to go into PS4 Mode, the controller should remember if the "Switch Touchpad and Share" setting is enabled.

### Current Behavior

When holding B4 upon plugging in to go into PS4 Mode, the controller does not initially remember to switch Touchpad and Share when the setting is enabled and must be unplugged and replugged for the setting to take effect again. Note that this issue does not occur when using the Web Configurator to change input modes to PS4.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Enable the "Switch Touchpad and Share" setting in the Web Configurator while in PS4 Mode.
2. Change input modes to anything besides PS4 Mode.
3. Unplug controller and change input modes back to PS4 Mode by holding B4 upon plugging in, and notice that Touchpad and Share are not swapped.
4. Unplug and replug the controller, and notice that Touchpad and Share are swapped as intended.

TheTrainGoes, 2023-06-18:
Hi @jazonandes ,

Thank you for the report - we will take a look into this when we have some time.

bsstephan, 2023-06-18:
I cannot reproduce this problem on my branch. Steps I took:

1. Web configurator: board was already in PS4 mode, so I unchecked and rechecked "Switch Touchpad and Share" (to make sure it was on), then selected PS3/DirectInput mode, and clicked Save
2. Confirmed via the dev console that what was sent to the board was "inputMode: 2, switchTpShareForDs4: 1"
3. Unplugged controller
4. Plugged in holding B4
5. Opened gamepad tester
6. Pressed S1, got Button 14 in the tester (correct switched behavior)
7. Pressed A2, got Button 9 (ditto)
8. Unplugged controller and plugged it back in
9. Repeated S1/A2 test, same results
10. Unplugged controller and plugged it back in holding B1 (PS3/DInput)
11. Pressed S1, got Button 9, pressed A2, got button 14 (correct unswitched behavior)
12. Unplugged controller and plugged it back in holding B4 (PS4)
13. Pressed S1, got Button 14, pressed A2, got button 9 (correct switched behavior)

My branch is main + a combination of hotkey and 4-way support, but IIRC neither branch changes anything material with the TP/Share switch. It's possible 7aa176e47ba8d5fc05c06c3386a0781e5384d87d (in main, not in a release yet) fixed some undiscovered logic issue with the TP/Share support.

jazonandes, 2023-06-22:
Hi @bsstephan, I believe you are correct. Just built from main and was unable to recreate the issue, then went back to the v0.7.2 release and the incorrect switching behavior occurred again.

TheTrainGoes, 2023-06-30:
Closing this issue as it will be resolved in the next reslease.

closed 2023-06-30 by TheTrainGoes