OpenStickCommunity/GP2040-CE #347: I2C block sharing between cores

dogtopus, 2023-06-18:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

When writing a GP2040 addon for one of my own I2C input peripheral, I noticed that my peripheral would constantly glitch out. A protocol capture shows that the data exchange between the OLED display and the peripheral are interleaved and corrupted.

This kind of setup is also allowed in some of the official addons like ADS1219 without (IIRC) explicit warning in the documentation so I'd presume that they would be broken as well.

* Firmware Version: 7aa176e
* Name of device: Any
* Link to where you purchased the device: Any
* Is this a custom build?: Yes
* Operating System: Any
* Browser: Any

### Expected Behavior

I2C block is properly shared without corrupting the data. 

### Current Behavior

Data corruption occurs when the same block is used by 2 cores.

### Steps to Reproduce

1. Enable OLED display
2. Enable some other addon that uses I2C and select the same block and pins the OLED display uses
3. Connect all devices to a single bus
4. Watch it crash and burn

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

Fortinbra, 2023-06-18:
Thanks for making a ticket for this! We have seen this while working on other I²C add-ons, and some work has been done to alleviate the issue, with the temporary work around to not enable the OLED if your using the other I²C add-ons. 

That said, we are looking for ways to make all the I²C devices play nicely with each other. We've discussed a few amongst the regular contributors, but no final answers just yet.

mikepparks, 2023-06-18:
The OLED is fairly chatty, as you have mentioned. Currently investigating options to improve our utilization of i²c to allow addons to play nice together. As more addons that use it get created, we need to manage the traffic better.

asturur, 2023-12-19:
so i was trying to get some insight on how to attach an i2c nfc reader to the gp2040-ce firmware. Would it be safe to use the i2c1 block block on different pins  on the same core 0?
I m new to everything here and i would like to avoid bugs i would never be able to distinguish from my own errors.

closed 2026-10-11 by arntsonl