OpenStickCommunity/GP2040-CE #364: build error

chiyu521, 2023-06-22:
[ 80%] Building CXX object CMakeFiles/GP2040-CE.dir/src/configs/webconfig.cpp.obj
In file included from /home/chiyu/pico/GP2040-CE/headers/gamepad.h:18,
                 from /home/chiyu/pico/GP2040-CE/headers/storagemanager.h:15,
                 from /home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:4:
/home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp: In function 'DataAndStatusCode setConfig()':
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:740:387: error: no match for 'operator=' (operand types are '_Config' and '<brace-enclosed initializer list>')
  740 | _init_default, false, LEDOptions_init_default, false, AnimationOptions_Proto_init_default, false, AddonOptions_init_default, false, ForcedSetupOptions_init_default}
      |                                                                                                                                                                    ^

/home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:1266:18: note: in expansion of macro 'Config_init_default'
 1266 |  *config.get() = Config_init_default;
      |                  ^~~~~~~~~~~~~~~~~~~
In file included from /home/chiyu/pico/GP2040-CE/headers/gamepad.h:18,
                 from /home/chiyu/pico/GP2040-CE/headers/storagemanager.h:15,
                 from /home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:4:
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note: candidate: 'constexpr _Config& _Config::operator=(const _Config&)'
  658 | typedef struct _Config {
      |                ^~~~~~~
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const _Config&'
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note: candidate: 'constexpr _Config& _Config::operator=(_Config&&)'
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to '_Config&&'
make[2]: *** [CMakeFiles/GP2040-CE.dir/build.make:245：CMakeFiles/GP2040-CE.dir/src/configs/webconfig.cpp.obj] error 1
make[1]: *** [CMakeFiles/Makefile2:1845：CMakeFiles/GP2040-CE.dir/all] error 2
make: *** [Makefile:136：all] error 2

TheTrainGoes, 2023-06-22:
Hi @chiyu521 ,

Can you give us some more info like:
- What operating system are you on
- Have you ever built successfully before
- What troubleshooting steps you have done already

chiyu521, 2023-06-22:
ubuntu 22.04
Previously it was possible to build successfully
Can't build after git pull

TheTrainGoes, 2023-06-22:
Hi @chiyu521 ,

Can you try fully removing your working directory and then doing a full pull of the repo?

chiyu521, 2023-06-22:
ok i'll try

chiyu521, 2023-06-22:
Fault persists

[ 80%] Building CXX object CMakeFiles/GP2040-CE.dir/src/config_utils.cpp.obj
[ 80%] Building CXX object CMakeFiles/GP2040-CE.dir/src/configs/webconfig.cpp.obj
In file included from /home/chiyu/pico/GP2040-CE/headers/gamepad.h:18,
                 from /home/chiyu/pico/GP2040-CE/headers/storagemanager.h:15,
                 from /home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:4:
/home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp: In function 'DataAndStatusCode setConfig()':
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:740:387: error: no match for 'operator=' (operand types are '_Config' and '<brace-enclosed initializer list>')
  740 | lse, AddonOptions_init_default, false, ForcedSetupOptions_init_default}
      |                                                                       ^

/home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:1266:18: note: in expansion of macro 'Config_init_default'
 1266 |  *config.get() = Config_init_default;
      |                  ^~~~~~~~~~~~~~~~~~~
In file included from /home/chiyu/pico/GP2040-CE/headers/gamepad.h:18,
                 from /home/chiyu/pico/GP2040-CE/headers/storagemanager.h:15,
                 from /home/chiyu/pico/GP2040-CE/src/configs/webconfig.cpp:4:
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note: candidate: 'constexpr _Config& _Config::operator=(const _Config&)'
  658 | typedef struct _Config {
      |                ^~~~~~~
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const _Config&'
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note: candidate: 'constexpr _Config& _Config::operator=(_Config&&)'
/home/chiyu/pico/GP2040-CE/build/proto/config.pb.h:658:16: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to '_Config&&'

chiyu521, 2023-06-22:
Build PICO directly without modifying any code

mthiesen, 2023-06-23:
Can you try to replace the line `*config.get() = Config_init_default;` with `*config.get() = Config Config_init_default;`?

chiyu521, 2023-06-23:
> 您可以尝试用 替换该行`*config.get() = Config_init_default;`吗`*config.get() = Config Config_init_default;`？

repaired successfully

closed 2023-06-23 by chiyu521