OpenStickCommunity/GP2040-CE #365: Add extra F keys to keyboard mode. F13 and beyond

levigermina, 2023-06-22:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

Would be really useful to use the controller as a macro keyboard when not using it for gaming.

### Describe the solution you would like to see

Add the option to map f13 to f24 on the keyboard mode. (Hex code: 0x68 to 0x73)

### Can you submit a pull request?

No

### Are you planning on working on this feature yourself?
No, but I have a code snippet that can be added to www/src/Data/Keyboard.js that should add these keycodes?

`      
{ label: "F13", value: 0x68 },
{ label: "F14", value: 0x69 },
{ label: "F15", value: 0x6a },
{ label: "F16", value: 0x6b },
{ label: "F17", value: 0x6c },
{ label: "F18", value: 0x6d },
{ label: "F19", value: 0x6e },
{ label: "F20", value: 0x6f },
{ label: "F21", value: 0x70 },
{ label: "F22", value: 0x71 },
{ label: "F23", value: 0x72 },
{ label: "F24", value: 0x73 }
`

TheTrainGoes, 2023-06-22:
This seems like a good bounty.  Are you willing to sponsor the work for it?

arntsonl, 2023-06-28:
Debug got it done!

closed 2023-06-28 by arntsonl