OpenStickCommunity/GP2040-CE #371: YD-RP2040's pin 26 and 27 not working properly in version 0.7.2

luisvoo, 2023-06-24:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ✓] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ✓] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ ✓] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ✓] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.2
* Name of device: YD-RP2040
* Link to where you purchased the device: [Here](https://www.aliexpress.com/item/1005004004120604.html?spm=a2g0o.productlist.main.1.159c6ea5Fq9zdc&algo_pvid=62556f40-f016-4417-84c3-c042f8f814c2&algo_exp_id=62556f40-f016-4417-84c3-c042f8f814c2-0&pdp_npi=3%40dis%21SGD%214.63%214.63%21%21%21%21%21%402102196716875742301764392d077d%2112000029278724340%21sea%21SG%211789607579&curPageLogUid=yqKhfcfxKJfD)
* Is this a custom build?: Yes
* Operating System: Windows 10
* Browser: Google Chrome

### Expected Behavior

Pin 26 and Pin 27 (Both analog GPIOs) to be functioning when connected

### Current Behavior

What is the current behaviour?
Pin 26 and 27 are not functioning when connected, and stucks at HIGH after A1 is pressed

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Flash 0.7.2 Firmware to any YD-RP2040
2. Assign pin 26 and pin 27 to a button
3. Try pressing the buttons assigned to pin 26 and 27, then press A1

### Additional comments

Things I have tried :
1. Tried other YD-RP2040 microcontrollers
2. Shorting AGND to GND
3.  Rolling back to previous versions, this is issue is **not faced in 0.6.3**

Any help is appreciated, thank you!

TheTrainGoes, 2023-06-30:
Hi @luisvoo ,

Have you tested this on a proper Raspberry Pi Pico unit?  

Generally we do not support units from AliExpress or similar.

TheTrainGoes, 2023-07-04:
We are going to close this request as we have not heard back and do not support issues from clone units.  

If this can be replicated on a Pico please let us know!

closed 2023-07-04 by TheTrainGoes