OpenStickCommunity/GP2040-CE #382: Adding additional GPIO pins for Keybaord/Arcade/MAME

SpoonyAllen, 2023-06-30:
I only came across the project about an hour ago and I'm impressed with how well everything is put together. bravo to everyone involved!

I do have a small feature request if possible. I downloaded this with the intention of using a pi pico as a mame encoder and seeing how well the NKRO and HID boot descriptor code worked (better than circuitpythons standard implementation). So far all of that works amazingly well but i did notice you can only map 18 inputs out of the possibly 25 pins (including hijacking the pico's GP0 and GP1).

As it stands, this is a great solution for a single player setup but it leaves it a little lacking for a two player setup. Without adding a start and coin up button, each player is relegated 4 directional inputs and 3 input buttons plus start and coin for each player.  

Perhaps this can be expanded with an addon or something to that effect? Regardless, huge thanks again to everyone involved in this project.

TheTrainGoes, 2023-06-30:
Hi @SpoonyAllen ,

if I am reading this correctly you are just looking at the Raspberry Pu Pico and not one of the many projects that people have made which breakout additional pins from the RP2040.   

I would encourage you to checkout our RP2040 Advanced Breakout board which breaks out all but one of the pins from the RP2040 and may be a good fit for your needs.

SpoonyAllen, 2023-06-30:
@TheTrainGoes 

What I'm saying is that the keyboard configuration panel limits you to 18 inputs. While there are 25 you can use on a standard pico to create a fully featured keyboard encoder to achieve 2 players, 6 buttons, dedicated start-coin and at least one utility panel.

Hope that explains better. :)

TheTrainGoes, 2023-06-30:
Hi @SpoonyAllen ,

Yes I understand now.   This should be doable.  

We normally take requests like this and have them sponsored as bounties.  

If you are willing to toss a few bucks into the pool for it let me know otherwise I will cover it.

SpoonyAllen, 2023-06-30:
I can definitely contribute to the pool. I don't know how to do it yet but as soon as i can, absolutely. 

As of right now, this project is the only one that supports keyboard, NKRO and hybrid keyboard. There is another, far more simplistic keyboard project aimed at arcade use but its based on circuitpython which still has HID descriptor issues.

InfraredAces, 2023-06-30:
Join the [OpenstickCommunity Discord](https://discord.gg/k2pxhke7q8) and look at #bounty-board or #bounty-chat

TheTrainGoes, 2023-07-21:
Looking at this again I believe the best approach is to have something added to the keyboard-mapping page in the web-config that lets you add additional undefined pins and map them to letters.

SpoonyAllen, 2023-07-31:
> Looking at this again I believe the best approach is to have something added to the keyboard-mapping page in the web-config that lets you add additional undefined pins and map them to letters.

lol i'm not picky. I just feel with this addition, the GP2040-CE project would be an excellent solution for everyone, including Arcade/mame fans. Having an inexpensive keyboard encoder on par with the basic functionality of the Ultimarc Ipac is desperately needed.

n-i-x, 2023-10-15:
I'm willing to sponsor a $200 bounty for anyone who can implement USB Composite into GP2040-CE and add multiple controller support. This would easily accommodate this PR as individual players could be exposed as HID gamepad, Keyboards, etc. and can be configured with any supported pin configuration.

Let me know if you want another PR for this request, or if this one is okay to redefine.

TheTrainGoes, 2025-05-07:
Going to close out this issue as we have added more GPIO pin assignments to keyboard mode.

@n-i-x I you may have seen it already but we have the start of USB host now.