OpenStickCommunity/GP2040-CE #390: Flatbox bricked

DasMoks, 2023-07-04:
Hey guys, just got a new Flatbox that had GP2040-CE installed. Tried to update the firmware and supposedly chose the wrong .uf2 (rev5 when i should have gone rev4 i guess) and kinda bricked my device... Buttons are shifted all around and I can't get into the web configurator to undo my damage... Have any idea how I can save that? Can't access web configurator or Bootsel...
Thanks in advance
Max

SpoonyAllen, 2023-07-04:
download the flash nuke from the downloads section 
https://gp2040-ce.info/#/download

Hold down the BOOTSEL button on your device and plug it in.

Drag the Nuke file into your device and it should erase your old firmware and allow you to start over.

DasMoks, 2023-07-04:
hey Spoony! Thanks for your reply, I already was able to nuke that, but I can't figure out a firmware that is working... Any ideas? I messaged the seller on Ali to find out which firmware they used, but I am kinda pessimistic about that approach,..

closed 2023-07-04 by DasMoks