OpenStickCommunity/GP2040-CE #391: Used Pins Bad Reporting in Web Config (Dev/'main' branch)

arntsonl, 2023-07-04:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 'main' latest as of 07/04/2023
* Name of device: Pico
* Link to where you purchased the device:
* Is this a custom build?:
* Operating System:
* Browser:

### Expected Behavior

Used pins should not be reporting incorrectly on the web config

### Current Behavior

Used pins in the add-ons page are reporting incorrectly

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Enable the Analog Add-on
2. Set Analog 1 ADC to pin 26 and pin 27 respectively
3. Save the Add-on page and reload
4. Set Any other Analog option
5. Observe the "pin 26" and "pin 27" options report an error with previously defined pins

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

[image]

Pelsin, 2023-07-05:
I think the issue is that these pins are used somewhere else but still show up as options in addons page.
This should hide non available pins: https://github.com/OpenStickCommunity/GP2040-CE/pull/392

arntsonl, 2023-07-19:
I'm going to close this issue since we fixed this for 0.7.3 and just make a new one as needed.

closed 2023-07-19 by arntsonl