OpenStickCommunity/GP2040-CE #398: PicoFightingBoard : Turbo button don't work

zeyingzheng-linux, 2023-07-07:
### Environment
* Firmware Version: v0.7.2 and I add some code by myself to enable turbo button
* Name of device: PicoFightingBoard 
* Is this a custom build?: Yes , I add some code by myself  to enable TURBO button
* Operating System: Ubuntu 20.04 

### Expected Behavior

Turbo button can work well in PicoFightingBoard 

### Current Behavior

Turbo button didn't work

### The code I add
[image]

### My Build command
[image]

### I add my code base on this commit 
[image]

bsstephan, 2023-07-07:
Hello. You changed the right stuff, but you don't need to change the code to enable turbo. The .h file contains the default settings for the board. You should go to the web configurator turbo addon and enable it there. If the turbo addon configurator doesn't work for some reason, please let us know.

closed 2023-07-07 by bsstephan

TheTrainGoes, 2023-07-07:
Hi @zeyingzheng-linux ,

May I ask why you compiled this for Turbo when it can just be added in web-config?

zeyingzheng-linux, 2023-07-08:
> Hi @zeyingzheng-linux ,
> 
> May I ask why you compiled this for Turbo when it can just be added in web-config?

Thanks for quick respond , I compiled the code the enable Turbo button just for experimention .
I don't read the whole source code , So I don't know what method webconfig will save the config .
Either filesystem or flash , if so ,I don't want this configuration to remain in the board all the time unless I use the webconfig to remove it .
In other words, I concern that the webconfig will remain in the board  even if I reflash the *.uf2 image , So I don't use the webconfig before.

According to your answer , I know maybe webconfig will cover my modification. I will read the webconfig code for my experimention , thanks .

bsstephan, 2023-07-09:
@zeyingzheng-linux if it saves you any time, the configuration is saved at the very end of the flash memory, so you wouldn't overwrite it with a .uf2 unless you were doing something unusual.

zeyingzheng-linux, 2023-07-09:
> @zeyingzheng-linux if it saves you any time, the configuration is saved at the very end of the flash memory, so you wouldn't overwrite it with a .uf2 unless you were doing something unusual.

@bsstephan Thanks for your answer. According to your recommendation , I try the web-config and it  can enable the turbo button, but the configuration will be clear when I reflash my .uf2 . Will it clear the configuration every time .uf2 is flushed ?
If so , Why does gp2040-ce.info recommended to flash flash_nuke.uf2 , then flash my .uf2 image ，Is this necessary ? 
That is the reson I don't use the web-config before, because I don't know the whole mechanism and I don't have any debug tool for my board, So I just read some code about turbo button then modify them, finally, I will check whether the turbo behavior meet my expectations.
[image]

By the way , I overwirte some variable in turbo.cpp then I can enable turbo button in default without web-config .

[image]

But I want to achieve the same goal with below code , But it can't not work , I just want to find out a easy way to change the default configuration without web-config . Could you give some advice ?

 
[image]
[image]

bsstephan, 2023-07-09:
@zeyingzheng-linux the configuration will *not* get lost when you flash a new .uf2. The .uf2 file only affects the firmware code, the configuration lives in a different section of flash. The recommendation to use flash_nuke.uf2 is only for troubleshooting, or if your board was used for a different project, in order to ensure the config is not corrupt.

I would advise not fighting against the current code, because setting it via webconfig is stable, but if you want to have the Turbo enabled by default, your first commit changing BoardConfig.h is a way to do it.

TheTrainGoes, 2023-07-09:
@zeyingzheng-linux I would also like to point out that if the goal of this is to have turbo persist through a reboot you can check out SHMUP mode in web-config which has exactly this.

zeyingzheng-linux, 2023-07-09:
> @zeyingzheng-linux the configuration will _not_ get lost when you flash a new .uf2. The .uf2 file only affects the firmware code, the configuration lives in a different section of flash. The [...]

@bsstephan I try that , if I just modify confiuration in BoardConfig.h , it couldn't take effect . I must change the configurtion in BoardConfig.h and add some code to overwire the variable in turbo.cpp then turbo button work fine . But I think it is not a good way . I confuse that why so hard to change the default configuration without webconfig . For example , I want to change the default vaule of turbo button, I need to read the implementation of turbo button , then I find out the overwrite method. I think there maybe a simple way to change the default configuration without web-config.

bsstephan, 2023-07-10:
Once you have the configuration generated, which is basically once you've booted the board once, the board defaults are no longer used. Basically, BoardConfig.h is what is used to populate an empty configuration --- once a configuration is set on the board, it always overrides the defaults.

If you would like to avoid changing BoardConfig.h but nevertheless configure a board with turbo enabled, I am working on tools in https://github.com/OpenStickCommunity/gp2040ce-binary-tools to create images with configuration overrides included, but currently the only way to change the defaults is to change BoardConfig.h.

zeyingzheng-linux, 2023-07-11:
> Once you have the configuration generated, which is basically once you've booted the board once, the board defaults are no longer used. Basically, BoardConfig.h is what is used to populate an empty [...]

Thanks for you answer , I think mabey picotool is useful for me . But I have another question , Once I dump the whole board image  use the command "picotool save -a whole-board.bin" . How can I flash that bin to other board ? Does picotool  provide such a function ? Once I flash that whole-board.bin to the new board , Can the new board get the whole function the same as orignal board ?