OpenStickCommunity/GP2040-CE #400: Trying to make USB Keyboard Host Work on RP2040

ez2torta, 2023-07-08:
### Prerequisites

Please check the following before posting an issue / bug report.

I tried wiring a USB Female port using this schematic

https://user-images.githubusercontent.com/77402236/241382084-8d2a5936-78b6-46ed-bc74-0ec21d629bd6.png

and compiled the latest main branch https://github.com/OpenStickCommunity/GP2040-CE/commit/25604fe4b62de2693724f14817935081da60049f

not working unless i'm missing something :/

- [ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 25604fe4b62de2693724f14817935081da60049f
* Name of device: Raspberry Pi Pico
* Link to where you purchased the device:
* Is this a custom build?: Yes
* Operating System: Linux
* Browser:

### Expected Behavior

I expect the USB Keyboard Host to work before and after RP2040 boot sequence

### Current Behavior

What is the current behaviour?

the keyboard host does not work, if i plug a keyboard before turning on the RP2040 the controller does not show. If i plug the keyboard after the RP2040 is working as a xbox360 controller also does not work.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Wire the female USB as the diagram above
2. Flash the current main branch 
3. use the webconfig for the D+ and D- pins
[image]
4. Keyboard not recognized. board not working if usb device is plugged

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

arntsonl, 2023-07-08:
Which keyboard are you using? I found that some keyboards with non-generic HIDs do not work such as Razer, while others work fine.

ez2torta, 2023-07-08:
I tried a taser keyboard but also a HP keyboard and a Kensington keyboard and none of them worked. 

I'm kinda worried that the rp2040 does not boot correctly when I have the usb keyboard connected. Is that expected?

ez2torta, 2023-07-19:
Now tried with the latest RC firmware and same results.. should i use certain pins for the D+ and D- wires?

TheTrainGoes, 2023-07-21:
Hi @ez2torta ,

Due to some ongoing concerns around USB host mode we have removed it from the current build.  You are welcome to continue exploring it on your own or join up on the Discord and see if there is anyone else that can try to assist.

closed 2023-07-21 by TheTrainGoes