OpenStickCommunity/GP2040-CE #402: ability to remap "twin stick" oled display to right analog output

siegarettes, 2023-07-08:
Currently the Twin Stick OLED display profile is set up to respond to the dpad and face buttons, as per the standard mappings for Virtual On controllers. So the dpad controls the left stick display, while the right stick is play responds to the North, East, West and South face buttons as directions (ex. triangle is up, square is left, circle is right, cross is down). 

For my current twin stick setup, I have the left stick wired as dpad, emulating the left analog stick, and the right stick as dual analog directions, emulating the right stick. So when displaying the twin stick layout the left stick will respond, but the right stick will not. It WILL respond to the face buttons as expected, however. 

### Describe alternatives you've considered

This could be solved by wiring the right stick as the face buttons, but it would lose a lot of functionalilty across games just to display inputs. 

### Can you submit a pull request?
No.
### Are you planning on working on this feature yourself?
No.

bsstephan, 2023-07-08:
Based on this and the conversation in #support, I have a suspicion that the problem here is the LED `process()` is checking gamepad state before DDI `process()` is contributing to the state. I've been wondering if we need to give each addon some kind of priority so that we can have them chain together in a particular order.

TheTrainGoes, 2023-07-09:
Hi @siegarettes,

The main issue here is that we currently do not have a way to differentiate DPad / LS / RS on the screen.  They are all referred to under the same set of defines.  This is why I did the Twinstick layout the way I did it originally.  

In order for this to be done we would need to redo all of the display code to allow for both sticks to be displayed separately.  I am also unsure how this change would work with the normal layout when you change from DPad to LS or RS.

Overall it seems like a simple change, but this will require quite a bit of behind the scenes work.  

We can put it up as a bounty if you like and see if anyone takes it on.

TheTrainGoes, 2023-07-21:
There was no response to this so this issue will likely sit for a while.

TheTrainGoes, 2025-05-22:
This work was completed and you can now display analog on the display.

closed 2025-05-22 by TheTrainGoes