OpenStickCommunity/GP2040-CE #407: Adding animated GIF support to I2C display splash mode

Quendi6, 2023-07-12:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?
I am very frustrated that we can't be able to upload and display animated GIF on I2C in splash mode.

### Describe the solution you would like to see

Implement animated GIF support for I2C display splash mode.

### Describe alternatives you've considered

I was thinking about create a branch with https://github.com/bitbank2/AnimatedGIF/blob/master/src/AnimatedGIF.h implemented and compile my own version.

### Can you submit a pull request?

Unfortunately no, I don't have enough free time to work on this.

### Are you planning on working on this feature yourself?

Maybe in the future, but it will take a long time.

TheTrainGoes, 2023-07-21:
Hi @Quendi6 ,

I find the ton a bit off in the request but agree that it would be nice to have .gif support.  

I would like to remind you that we are a fully open source project that people contribute to in their spare time.  

If you are willing to sponsor this work we can add it to our bounty wall and see if anyone would like to pickup the work.

Quendi6, 2023-07-22:
The ton is as it was in the request template... Even though English is not my native language, I find it a bit funny too, but who am I to judge and not show respect to the author of the model?

I promise if I work on it it will be shared here with everyone.

HoboDev, 2023-08-11:
Something that could work in a similar fashion is sprite animation. Which might be easier to implement and have a smaller footprint.

TheTrainGoes, 2025-01-03:
Closing this as we have no plans to add gif support with the current space limits.

closed 2025-01-03 by TheTrainGoes