OpenStickCommunity/GP2040-CE #410: TURBO functions not fully working

fostrike, 2023-07-14:
### Prerequisites

Please check the following before posting an issue / bug report.

- [yes] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [yes] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [yes] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [yes] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* I'm using an RP2040-Zero, testing on a breadbord.

### Expected Behavior
Basic TURBO function is working OK, but seems to be capped a 3 shots/second.
Two functions available in the web config, Turbo Shot Count and Turbo Dial doesen't seems to work.

### Current Behavior
-"Turbo Shot Count" - Doesent change the frequency of the shots, as said before seems capped a 3Hz.
-"Turbo Dial" - I connected a 10K potentiometer on GP29 (after changing the default pin assignment) but doesn't seems to do any change in the shoots frequency.

TheTrainGoes, 2023-07-14:
Hey @fostrike,

Are you connecting the pot to an ADC pin?

fostrike, 2023-07-14:
Yes, pin 29 but there is no way to make mistakes, the web interface force you to pick only the 4 analog pins available.

arntsonl, 2023-07-19:
For the potentiometer, you have to use an ADC pin. Can you try GP26-GP28? There is a chance it doesn't work on GP29 on the rp2040 zero.

Turbo button itself is working fine on the latest build.

fostrike, 2023-07-20:
OK Arntsonl, you are totally right, if I use the default port pin 26 as analog input the DIAL function works fine and the actual turbo value can be see in real time on the OLED.
I'm using the Waveshare RP2040-Zero, that's why i changed the ADC default pin.
Thank you!
By the way, yesterday was looking at the turbo.cpp code, I'm not a programmer so... but this line seems strange..
adcShmupDial = 26 - options.shmupDialPin;

TheTrainGoes, 2023-07-21:
Closing as resolved

closed 2023-07-21 by TheTrainGoes