OpenStickCommunity/GP2040-CE #411: The TURBO LED does not work properly

seunggoon, 2023-07-15:
I am having some problems using GP2040-CE and need to ask for help.
I don't know how to connect TURBO LED.
When I set LED Positive to 3.3v, LED Negative to GPIO:
Nothing set to turbo: LED off
Button set to turbo but not being pressed: LED off
Button set to turbo and is pressed: LED blinks on

LED Positive to GPIO Pin, LED Negative to GND:
Nothing set to turbo: LED on
Button set to turbo but not being pressed: LED on
Button set to turbo and is pressed: LED flashes on and off

Pins used:
Turbo = GPIO 14
Turbo LED = GPIO 15

I don't think that's a normal behavior,
The normal effect should be:
Nothing set to turbo: LED off
Button set to turbo but not being pressed: LED on
Button set to turbo and is pressed: LED blinks on
How to connect Turbo LED in general?

GP2040-CE 0.7.2

TheTrainGoes, 2023-07-15:
Hi @seunggoon ,

Your current situation is correct.  The LED will only bill when the button that has been assigned to turbo is active (blinking).   

It was not designed to be on all of the time when any of the Turbo buttons are active.

seunggoon, 2023-07-15:
@TheTrainGoes
First of all, thank you for your answer, but I am still not sure how to connect the LED. May I ask which pin should be connected to the positive pole of the TURBO LED? Which pin should the negative electrode be connected to?

TheTrainGoes, 2023-07-15:
Hi @seunggoon ,

As you stated here - this is the intended connection:

```When I set LED Positive to 3.3v, LED Negative to GPIO:
Nothing set to turbo: LED off
Button set to turbo but not being pressed: LED off
Button set to turbo and is pressed: LED blinks on

```

seunggoon, 2023-07-15:
Hi @TheTrainGoes，
Okay, thank you for your enthusiasm。

TheTrainGoes, 2023-07-21:
Closing as non-issue

closed 2023-07-21 by TheTrainGoes