OpenStickCommunity/GP2040-CE #423: Bad flash from Linux

Soundtoxin, 2023-07-23:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.3
* Name of device: Flatbox rev 5
* Link to where you purchased the device: Built from various parts myself following the guide
* Is this a custom build?: Yes
* Operating System: Guix System GNU/Linux
* Browser: Qutebrowser

### Expected Behavior

- mount microcontroller in bootsel mode
- copy .uf2 file to storage
- unmount
- controller works

### Current Behavior

- mount mcu in bootsel mode
- copy .uf2 file to storage
- unmount
- controller does not work, can't get web configurator or bootsel shortcut to work on GNU/Linux or macOS after flashing

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Hold Start+Select+Up and insert cable connecting to computer
2. `sudo tail /var/log/messages`, note /dev/sdX used by new drive
3. `sudo mount /dev/sdc1 /mnt/usb -o umask=000` (allows non-root users to write to the drive)
4. `ranger /mnt/usb ~/Downloads`
5. find .uf2 file, yy to copy, tab back to /mnt/usb, pp to paste
6. `sudo umount /dev/sdc`

I was going to try redoing it with rsync in case something was weird with ranger but haven't been able to get it to show up as storage again so far. Going to test on a Windows PC in a bit as I tend to have better luck flashing there for whatever reason. I have flashed many versions of GP2040-CE before to this and other Flatboxes but at some point it stopped working from my main machine for some reason. I don't have auto mount/unmount like I would on the Windows PC so I wonder if something weird is happening with that. Maybe some sync operation is happening that actually causes issues when I unmount.

bsstephan, 2023-07-24:
Is there a reason you don't just `cp` the file?

Soundtoxin, 2023-07-24:
I unscrewed the case to get at BOOTSEL directly and was able to flash it with Windows after that.
I have a one-liner in my shell history to mount sdc1 and open those commonly used dirs in ranger, so it's just more convenient that way. I'd probably use rsync over cp otherwise as it's more reliable and can resume interrupted transfers and such, but maybe cp would be fine here.

bsstephan, 2023-07-24:
`cp` simply works is my point. `ranger` and/or `rsync` might do something unexpected by BOOTSEL. for example the RP2040 reboots once writing is "over", so resuming an interrupted transfer wouldn't be possible anyway, and it probably doesn't like e.g. temp files.

In any event, we are not reaponsible for the BOOTSEL mode anyway (rather it's a core RP2040 boot ROM thing), so if it does something odd, there's nothing we can do.

Soundtoxin, 2023-07-24:
Yeah, this is maybe more of a Pico problem, but I thought someone here might have experience with this, and now next time it happens to someone, they'll hopefully find this issue. I will test updating a different Flatbox with cp later and report back.

bsstephan, 2023-07-24:
Sounds good. If you do have advanced use cases beyond `cp`, there is always picotool, or what I've been working on in https://github.com/OpenStickCommunity/gp2040ce-binary-tools

Soundtoxin, 2023-07-24:
Okay, I upgraded two in a row without issues using `cp`, so I guess that confirms that. Thank you for the help. Still did manual mount + unmount the same so I guess that wasn't the problem. Closing the issue now.

closed 2023-07-24 by Soundtoxin