OpenStickCommunity/GP2040-CE #431: Issue - Chase animation bug when increasing speed

InfraredAces, 2023-07-28:
### Prerequisites

Please check the following before posting an issue / bug report.

- [ X ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ X ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ X ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ X ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: [0.7.3](https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.3/GP2040-CE_0.7.3_FlatboxRev5RGB.uf2)
* Name of device: Flatbox Rev5 RGB
* Link to where you purchased the device: [OpenStickCommunity](https://discord.com/channels/1049366310389289001/1049371306958782495/1125457956775931956)
* Is this a custom build?: No
* Operating System: 6.4.1-arch2-1 (64-bit)
* Browser: Chrome

### Expected Behavior

Using the LED chase animation, repeatedly press S1 + S2 + R2, eventually reach a maximum speed where LEDs chase.

### Current Behavior

Using the LED chase animation, repeatedly press S1 + S2 + R2, eventually, the LEDs stop chasing and when pressed the Pressed LED color stays without disappearing.

Changing the LED animation works, but when going back to chase animation, above behavior continues.

Lowering the animation speed allows the chase animation to work once more, but only after changing LED animations away and then back to the chase animation.

### Steps to Reproduce

1. Plug in controller
2. Change the LED animation to Chase
3. From minimum speed, press S1 + S2 + R2 (LED Parameter Down) 32 times (will still happen if more than 32)
4. LED stop chasing
5. Press buttons and Pressed LED color will stay
6. Change the LED animation to another animation
7. Change the LED animation back to Chase
8. LED does not chase and pressed button LED color will remain
9. Lower chase animation speed by pressing S1 + S2 + R1 a few times
10. Change the LED animation to another animation
11. Change the LED animation back to Chase
12. LED does begin chasing again and pressed button LED color will no longer remain

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

[IMG_1076.webm](https://github.com/OpenStickCommunity/GP2040-CE/assets/44279698/9c05de6f-280f-4321-8c71-69bbb7c25c98)

arntsonl, 2023-07-28:
Yep its absolutely a bug! It crashes when the speed is too fast. Thanks for documenting, we'll get it fixed!

InfraredAces, 2023-09-02:
confirmed issue resolved with #474
closing issue

closed 2023-09-02 by InfraredAces