OpenStickCommunity/GP2040-CE #450: Flatbox firmware ignoring custom pin layout on most recent releases

MGStan0, 2023-08-08:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.3/0.7.2
* Name of device: FlatboxRev5
* Link to where you purchased the device: N/A
* Is this a custom build?: No
* Operating System: Windows 11
* Browser: Firefox

### Expected Behavior

The FlatboxRev5 Firmware should properly map the pin layout defined in the BoardConfig.h. e.g. START should be the left most tactile switch.

### Current Behavior

Pin layout is using the default GP2040-CE layout. Since the Flatbox does not use the default start button pin at all the web interface is inaccessible. Home is the leftmost tactile switch and START is missing.

### Steps to Reproduce

1. Put board into BOOTSEL mode
2. Load firmware  GP2040-CE_0.7.3_FlatboxRev5.uf2 or  GP2040-CE_0.7.2_FlatboxRev5.uf2 to the board
3. Restart board and test buttons

### Misc
The BoardConfig.h itself seems to properly define the layout but is being ignored for some reason.
I also tried compiling from source myself and the resulting GP2040-CE_0.7.3_FlatboxRev5.uf2 was still ignoring the custom layout.
Firmware 0.7.1 uses the custom layout properly.

TheTrainGoes, 2023-08-08:
Hi @MGStan0 ,

Can you perform a full Nuke and put the latest Rev 5 UF2 on and see if the issue persists?

MGStan0, 2023-08-08:
That seems to have done it. Now I'm really curious why 0.7.1 worked without nuking.

closed 2023-08-08 by MGStan0