OpenStickCommunity/GP2040-CE #461: OLED freezing when Passthrough is enabled without Dongle pluged in.

ReverendSID, 2023-08-19:
### Context

I'm using a Raspberry Pi Pico-H board, plugged into a screw terminal breakout board purchased through Amazon. https://www.amazon.com/dp/B094NLY2SN?psc=1&ref=ppx_yo2ov_dt_b_product_details  Arcade buttons, OLED display, also from Amazon https://www.amazon.com/dp/B094NLY2SN?psc=1&ref=ppx_yo2ov_dt_b_product_details and a self soldered USB plug for the passthrough, using a Magicboot PS4 v1.1 as the dongle.

* Firmware Version: 0.7.4
* Name of device: Pico-H
* Link to where you purchased the device: https://www.microcenter.com/product/650107/raspberry-pi-pico-h-raspberry-pi-pico-with-headers-pre-installed
* Is this a custom build?: Yes
* Operating System: Windows 10
* Browser: Firefox

### Expected Behavior

On 0.7.3 firmware I would plug the controller in while holding the appropriate button for the mode I wanted, the OLED would display the splash picture, and then switch the the stickless layout display with the input mode, and pressing buttons would lite those buttons on the display. I expected this behavior to continue on 0.7.4 when I turned on the Passthrough add-on.

### Current Behavior

When passthrough mode is enabled, but there is no dongle plugged into the passthrough, the OLED remains frozen on the initial image when the controller turns on. If splash screen is enabled in the display settings, it will never go off the splash screen. If splash screen is disabled it will show the controller layout screen, but will not show ongoing inputs. If I hold B2 to put it in XInput mode as I plug it in, the display loads with B2 illuminated, and never changes from that, same if I hold B4 for PS4, etc...

If I insert a dongle (in my case a Magicboot PS4) into the passthrough while the OLED is in this frozen state, it immediately returns to proper function.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. In Webconfig, enable Passthrough add-on.
2. Enable the Display, with or without splash screen.
3. Do NOT plug in a dongle to the USB passthrough.
4. Reboot board into controller mode.

At this point my screen is completely unresponsive, as described above. The controller functions just fine, all buttons work as they did on previous firmware. Just the screen is frozen, either on the splash, or the input display. Plugging the dongle in fixes this issue, and the screen functions normally again.

### Screenshots & Files

[image]

[image]

[image]

arntsonl, 2023-08-22:
@ReverendSID can you test RC2 and see if this fixes your issue?

ReverendSID, 2023-08-22:
> @ReverendSID can you test RC2 and see if this fixes your issue?

RC2 did fix the OLED issue for me.

closed 2023-08-24 by arntsonl