OpenStickCommunity/GP2040-CE #468: USB Keyboard Host Add-on can' t press more than 6 keys.

ez2torta, 2023-08-23:
### Prerequisites

Please check the following before posting an issue / bug report.

- [X] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [X] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [X] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [X] I am reporting the issue to the correct repository

### Context

* Firmware Version: 0.7.4
* Name of device: Raspberry Pi Pico

I flashed the firmware and configured the keyboard correctly.
I have it working on XInput and it recognized two of my keyboards perfectly! so i'm pretty happy with that.
I wanted to press more than 6 keys in my keyboard but the rest of the keys are not recognized.
It does not matter which keys are

### Expected Behavior

Please describe the behaviour you are expecting.

I press 6 keys in the keyboard and it works perfectly.
If i press a 7th key i expect it to be recognized as it would on a windows PC

### Current Behavior

What is the current behaviour?
I press 6 keys in the keyboard and it works perfectly.
If i press a 7th key then it does not get recognized.
I think that this is because the USB HID Report is bigger than the current implementation in TinyUSB.
(https://github.com/hathach/tinyusb/blob/1fdf29075d4e613eacfa881166015263797db0f6/src/class/hid/hid.h#L327) 
Maybe the keyboard sends 2 reports of 6 keys each? i dunno

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Have a USB Female Cable soldered into GP2040
2. Enable USB Keyboard Host in Webconfig
3. Go into controller mode and press 5 keys at the same time, should work fine
4. press the 6th key and then the 7th key
5. the 7th key should not be registered
6. then let go of the 6th and 7th and press the 7th key first and then the 6th one.
7. the 7th key should register but the 6th should not.

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

InfraredAces, 2023-08-23:
https://www.devever.net/~hl/usbnkro
TinyUSB sets connected devices to use boot protocol by default. Would need to parse HID report descriptors.

ez2torta, 2023-08-23:
> https://www.devever.net/~hl/usbnkro
> 
> TinyUSB sets connected devices to use boot protocol by default. Would need to parse HID report descriptors.

Super useful link! So the keyboards enter this bios mode by default which lead to the current implementation

mvergaraq, 2023-09-26:
Same issue, did you find a work around?

ez2torta, 2023-09-26:
> Same issue, did you find a workaround?

not for now. the current implementation of the USB keyboard input reading only allows for the BIOS compatible descriptor.

only if we have a way of trying to read the rest of the descriptor each keyboard outputs then we could press lots of keys :)

TheTrainGoes, 2023-10-08:
Hi @ez2torta ,

What is the use case for needing 7 keys pressed?

InfraredAces, 2023-10-08:
One use case I personally found was holding 3 directions (utilising SOCD cleaning) and holding 4 buttons. There are a number of fighting games where holding 4 buttons activates some in-game macro. 

Another would be for when the Tilt Add-on is integrated into the keyboard Host Add-On where you would often need to press multiple directions + modifiers + action buttons.

ez2torta, 2023-10-09:
> One use case I personally found was holding 3 directions (utilising SOCD cleaning) and holding 4 buttons. There are a number of fighting games where holding 4 buttons activates some in-game macro. [...]

Same, i was trying to do some SOCD tricks and sometimes the last input drops so it's kinda random

TheTrainGoes, 2025-03-13:
We are going to close this as `not planned`.  This is a limitation of the current implementation of Keyboard mode.  In order to change it we would need to change the entire mode.

closed 2025-03-13 by TheTrainGoes