OpenStickCommunity/GP2040-CE #498: Enabling Reverse function prevents the direction inputs from pin 16-19 from getting recognized

jchen2601, 2023-09-05:
1 - What device are you using?
TheTrain RP2040 Advanced Breakout Board version 5.2E
2 - Where did you buy it? (post link to sales page)
TheTrain from /r/fightsticks Discord
3 - What version of the firmware are you currently running? (check this via web-config)
v0.7.4
4 - Which distribution of the firmware are you currently running? (copy and paste the name of the firmware file you loaded on)
GP2040-CE_0.7.4_RP2040AdvancedBreakoutBoard.uf2
5 - What is the problem that you are having?
Enabling Reverse function to button on pin 23 disables all the other directional buttons that I bound to pin 16-19. How do I enable Reverse without disabling the other directional buttons?
6 - What trouble shooting steps have you already done?

Disabling Reverse allows the pin 16-19 directional buttons to be used again. I can bind additional regular directional input UDLR to the pin 23 button no problem. When I reenabled Reverse to pin 23, the pin 16-19 directional buttons get disabled again. Enabling Reverse for Left and Down input while disabling Up and Down input Reverse still disables the pin 16-19 directional buttons. Enabling Reverse for pin 29 also produces the same issue. I am connecting with Xinput. 

[image]

[image]

TheTrainGoes, 2023-10-08:
This has been fixed and merged into main via #499 .

Closing issue as resolved.

closed 2023-10-08 by TheTrainGoes

jchen2601, 2023-10-09:
How do I apply this fix? I downloaded the firmware for RP2040 Advanced Breakout from the GP2040-CE website and applied it to TheTrain breakout board. Enabling Reverse on pin 29 still disables directional buttons on pin 16-19. I can't assign directional button to other pins because they are connected via the 20pin harness or to the 16-19 pin of the screw terminal that corresponds to directions.

TheTrainGoes, 2023-10-09:
We have not put this to release yet.  It's been merged to main so you can compile yourself if you like otherwise you will need to wait until our next release is up.

jchen2601, 2023-10-10:
> We have not put this to release yet. It's been merged to main so you can compile yourself if you like otherwise you will need to wait until our next release is up.

Thank you. I will wait for the next release.