OpenStickCommunity/GP2040-CE #505: Colour on press for custom LED layout doesnt work in Left Stick or Right Stick mode

gingingin100, 2023-09-06:
### Prerequisites

Please check the following before posting an issue / bug report.

- [✓ ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [✓ ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [ ✓] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [✓ ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version:0.7.4
* Name of device:BentoBoxV1
* Link to where you purchased the device: https://scrubtier.co.uk/products/bentobox?variant=41780826931394
* Is this a custom build?:No?
* Operating System: Windows 11
* Browser: Opera GX

### Expected Behavior

Please describe the behaviour you are expecting.
 
When using a custom LED layout the LEDs under the Up Down Left and Right inputs should change colour according to what has been set in the configurator

### Current Behavior

What is the current behaviour?

This only occurs when in DPad mode not in Left Stick or Right Stick mode

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Create custom LED layout
2. Set a direction input to change colour when pressed
3. Change directions to correspond to a Left Stick or Right Stick
4. Actuate directions to see if the colour changes

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

TheTrainGoes, 2023-10-08:
We have some LED code rewrite going on so this may be taken care of by that. 

If that re-write addresses the issue we will close this out.

TheTrainGoes, 2026-11-05:
This has been resolved as part of the RGB refactor

closed 2026-11-05 by TheTrainGoes