OpenStickCommunity/GP2040-CE #506: Web Config, page: Settings, inconsistent vertical and horizontal distances in section settings.

HoboDev, 2023-09-07:
### Prerequisites

Please check the following before posting an issue / bug report.

- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.4 (6f5a8c0)
* Browser: Chromium & Firefox

### Description

The vertical distance between the 4-way joystick switch and the label "Profile Number" is not the same as for the other options. There are wrapping elements missing.

Current:
```
<div class="form-check form-switch">
  <input type="checkbox" id="fourWayMode" class="form-check-input">
  <label title="" for="fourWayMode" class="form-check-label">4-Way Joystick Mode</label>
</div>
```

Fixed:
```
<div class="row mb-3">
  <div class="col-sm-3">
    <div class="form-check form-switch">
      <input type="checkbox" id="fourWayMode" class="form-check-input">
      <label title="" for="fourWayMode" class="form-check-label">4-Way Joystick Mode</label>
    </div>
  </div>
</div>
```

Also the distance between border and input for "Profile Number" is smaller than the other inputs. (This might be fixed by converting the number input to a drop down, the wrapping elements are correct but the number input somehow is off)

### Screenshots & Files

[image]

mikepparks, 2023-09-07:
For what it's worth, this is good, albeit nitpicky, feedback that could have also been a PR.

HoboDev, 2023-09-11:
I'm not sure if this an invitation for a PR or the attempt to repulse someone interested in helping out.

If it was only the 4-Way Joystick Mode toggle I wouldn't have bothered but the Profile Number input is quite obvious, so why not tackle both? At least it's the user facing part which is of some significance and makes a good impression if kept tidy.

closed 2023-09-17 by HoboDev