OpenStickCommunity/GP2040-CE #515: Force Feedback Support?

tlalexander, 2023-09-12:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

I would like to build a racing wheel using this project which supports force feedback. Incidentally I am building this with my open source brushless motor controller: https://github.com/Twisted-Fields/rp2040-motor-controller

### Describe the solution you would like to see

I would like to be able to define a force feedback device with this project for a racing wheel, joystick, or other FFB device (not just rumble).

### Describe alternatives you've considered

I cannot find any RP2040 library which supports FFB.

### Can you submit a pull request?
It seems tricky to code this, but I can certainly submit a PR if I get some help writing the code.

### Are you planning on working on this feature yourself?

Ideally no, but perhaps I can find some time. I need guidance though.

tlalexander, 2023-09-12:
Here is an arduino library with force feedback support. I currently am not sure I have the time to learn enough about this stuff to do a proper integration, but perhaps it is helpful for someone:
https://github.com/sgtnoodle/ArduinoJoystickWithFFBLibrary

TheTrainGoes, 2023-10-08:
Hi @tlalexander ,

Currently we don't have anyone focused on force feedback / wheel implementation but that may change in the future if someone joins that wants to work on it or we get a PR for it.  

I will leave this open for now.

TheTrainGoes, 2023-12-10:
Hi @tlalexander ,

We spoke about this again internally and do not believe this will be in our roadmap anytime soon.  We will be closing this as not currently planned however anyone is free to work on it if they choose and submit a PR

closed 2023-12-10 by TheTrainGoes