OpenStickCommunity/GP2040-CE #519: Can not join discord

Joomaton, 2023-09-14:
[image]

Pelsin, 2023-09-14:
Hi! Sorry to hear you are having trouble joining the discord.
What link did use, was it https://discord.gg/k2pxhke7q8?

It might also be that a phone verification is required: https://support.discord.com/hc/en-us/articles/6181726888215-Verification-Required-FAQ

If it still does not work, try opening the invite link in an incognito window.

Joomaton, 2023-09-14:
Yes. That's the link I've used.
Phone verification would be odd. I never did that but I can chat on other channels without any issue.
I hope discord is not turning google.

TheTrainGoes, 2023-09-15:
Hi @Joomaton ,

There was a period of time recently where we had entrance screen set to the highest.  It has been lowered a little.  

If you are still unable to join then your account does not meet our criteria for entry.

closed 2023-09-15 by TheTrainGoes