OpenStickCommunity/GP2040-CE #532: 4-Way Joystick Mode behaves differently from expected

sirrow, 2023-09-20:
### Prerequisites
- [x] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [x] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [x] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [x] I am reporting the issue to the correct repository

### Context
* Firmware Version: v0.7.4
* Name of device: Pico (but the issue should be not depend to device.)
* Is this a custom build?: no

### Description of Issue

I found 4-Way Joystick Mode behaves differently from my expectation.
It seems for me that SOCD and 4-Way Joystick Mode seems conflicted and 4-Way Joystick Mode need to care only diagonal direction but also opposite direction.
It would be grateful if some of you consider if my expectation reasonable.
2 examples I feel strange are below.

#### Example 1
##### Preconditions
- SOCD is Last Win
- 4-Way Joystick Mode is enabled

##### Scenario
1. push left
2. push right
3. push down
4. release right

##### Expected Output
down is pressed

##### Actual Output
left is pressed because SOCD outputs left at step 4 and 4-Way Joystick Mode get output of SOCD.

#### Example 2
##### Preconditions
- SOCD is Off
- 4-Way Joystick Mode is enabled

##### Scenario
1. push left
2. push down
3. push right

##### Expected Output
right is pressed

##### Actual Output
Both left and right are pressed. Because 4-Way Joystick Mode only seems only checking which of directions which consists diagonal is pressed last and SOCD is Off.

sirrow, 2023-09-20:
Examples are checked in below environment:
- Input Mode in GP2040-CE
  - Xinput
- gamepad tester (reproduced in both web site)
  - https://hardwaretester.com/gamepad
  - https://pelsin.github.io/gamepad-tester/
- Browser
  - Chrome
- OS
  - Windows 11

bsstephan, 2023-09-20:
@sirrow - if you get a chance can you try my changes in https://github.com/OpenStickCommunity/GP2040-CE/compare/main...bsstephan:GP2040-CE:four-way-improvements ? I have 4-way running before SOCD now, and it makes Example 1 work as expected. I'm wondering if you can try it out and make sure it didn't break some other expectation, but it looks good to me. I do not know if it addresses Example 2 at all.

bsstephan, 2023-09-20:
Oh, found a new breakage, now Example 1 is fixed, but if you instead do Left, Down, Right you get... nothing. Wild. Well I'll mess with it more later.

bsstephan, 2023-09-20:
Okay. I think @sirrow's branch fixes my issue. Assuming @TheTrainGoes can confirm that #533 fixes the SOCD=Off case, my testing suggests we should merge both #533 and #537.

sirrow, 2023-09-21:
I tested code on my testing branch merging both #533 and #537.
With the branch, both examples got result as my expectation.
I cannot find any flaw about 4-Way Joystick Mode from test and logic point of view in case both https://github.com/OpenStickCommunity/GP2040-CE/pull/533 and https://github.com/OpenStickCommunity/GP2040-CE/pull/537 are merged.

PR #537 (2023-09-25): run 4-way filtering before SOCD cleaning
This addresses Example 1 of #532 without any immediately obvious detriment to the 4-way/SOCD interaction, other than it has an order of operations problem (L > D > R = nothing, but R > D > L = L).

However, #533 fixes that issue by implementing filtering better, and it fixes Example 2 of #532.

@TheTrainGoes is going to do more testing, but in my testing, these two PRs together fix #532.
closed 2023-09-25 by arntsonl