OpenStickCommunity/GP2040-CE #552: Also supports d-pad + left analog

parbzip, 2023-09-28:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

An unlock hotkey requires L2 + R2 + UP(D-PAD)
Menu Arrow keys (left analog)
In use, you need left analog to switch back and forth D-PAD
Can be used both simultaneously or via hotkey Pad Mode (left analog)
S1 + S2 + UP is a direct trigger UP (D-PAD)
The current function is "switch". Is it possible to add "trigger" directly?

### Describe the solution you would like to see

Provide a clear and concise description of what you want to happen.

### Describe alternatives you've considered

Let us know about other solutions you've tried or researched.

### Can you submit a pull request?

No.

### Are you planning on working on this feature yourself?

No.

TheTrainGoes, 2023-09-28:
Hi @parbzip ,

Currently you can change via shortcut or slide type switch.   That slide type switch can actually be two separate latching switches if needed.  You can also change the shortcut used to meet your needs better like having a function key assigned and using it in combination with another key.   

Does all of this not meet your needs?

parbzip, 2023-09-29:
I want a function to "trigger" the following keys
https://i.imgur.com/72WbqyH.jpg[](url)

TheTrainGoes, 2023-09-29:
Sorry @parbzip I am not following what you have in that image.  

Can you very clearly try to write out how you want this to work?

parbzip, 2023-09-29:
Hotkey Settings
add
1. Multiple identical hotkeys can be triggered at the same time
2. Hotkey can add D-PAD UP (similar to L3 R3 button)
========================================
Press the S1+S2+UP hotkey L2 R1 D-PAD UP to operate simultaneously

TheTrainGoes, 2023-09-29:
OK - this sounds more like you want macros.

TheTrainGoes, 2023-10-08:
I am going to close this request as we will be rolling out macros soon and you will be able to set this up there.

closed 2023-10-08 by TheTrainGoes