OpenStickCommunity/GP2040-CE #559: New 4-way implementation seems to conflict with itself when enabled for both normal and DDI directionals

bsstephan, 2023-10-05:
Pasting from ~~IRC~~ Discord, as is my tradition:

> I tested more and it is stranger than I thought. bear with me:
> defining the desired behavior as Down -> Down+Right = Right and Right -> Down+Right = Down...
> 1) if 4-Way is enabled, the desired behavior is present on the main directionals
> 2) if Dual Directional 4-Way is enabled *and normal 4-Way is disabled*, the desired behavior is present on the secondary directionals
> 3) if *both* 4-Ways are enabled, the behavior on both changes to Down -> Down+Right = Right and Right -> Down+Right = Right, giving preference to left/right (not up/down as I said initially)
> I don't immediately have an explanation for why this is, but my guess is something with the history is getting into a bad state if both sets of directions are contributing to the history?
> this is probably a very strange corner case and maybe only meaningful to someone like me testing a bunch of stuff after code changes

Writing this down for when myself or @sirrow can take a look.

sirrow, 2023-10-09:
I reproduced the issue.
Basically horizontal axis(left and right) wins vertical axis(up and down).

With my quich experimental patch https://github.com/sirrow/GP2040-CE/commit/52b03e431d062c64c156634796fc297c1b90b046, it seems better.

According to the result of the patch, sharing state for main input and for DDI input makes something bad result.
It seems good to stop "static bool inList[]" and "static list<DpadDirection> dpadList" in updateDpad() method. It seems better to give these these as argument of updateDpad(), or keep it in each private variable member for each class.

bsstephan, 2023-12-27:
Assigning this to myself so I remember to take a look at @sirrow's patch, because I keep forgetting.

> Even with my experimental patch, there is a behavior I do not clarify either good or not in below scenario.
> 
>     1. push MAIN LEFT
> 
>     2. push DDI DOWN
> 
>     3. got LEFT DOWN as output
> 
> 
> Is this output as to be? or output should be DOWN?

Am I right to guess that you have DDI Combine Mode set to "Mixed"? If so, I think the behavior you mention is an unfortunate corner case, but too specific of one to fix right now. DDI Mixed mode is probably going to go away soon as core starts consuming more of its features, so I'm inclined to just leave it for now.

I will look at your experimental patch when I get a chance.

PR #733 (2023-12-29): Fix 4-way mode when enabled for both core gamepad and DDI
This fixes the misbehavior described in #559 item 3, where 4-way didn't work properly if it was enabled for both the core gamepad and DDI at the same time. @sirrow is the author, I have just pulled, reviewed, and tested it.

Fixes #559.
closed 2023-12-29 by arntsonl