OpenStickCommunity/GP2040-CE #560: Proposal: remove Invert Y Axis from the default hotkeys

bsstephan, 2023-10-05:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

N/A

### Which problem is this feature request solving?

People seem to be more confused by accidentally hitting the invert y-axis shortcut than are happy it exists by default.

### Describe the solution you would like to see

Put something else (or nothing, that's fine too) in that hotkey slot going forward.

### Can you submit a pull request?

Yes but lol time. This might be a good first issue after there is consensus.

### Are you planning on working on this feature yourself?

Would love to, but it's way down on my personal priority list.

TheTrainGoes, 2023-10-09:
I'm all for removing it as a default shortcut.

TheTrainGoes, 2023-10-09:
I intended to make a PR for this but ended up just yoloing it to main.

TheTrainGoes, 2023-11-23:
Forgot to close this - the change has been made.

closed 2023-11-23 by TheTrainGoes