OpenStickCommunity/GP2040-CE #572: [Feature] Playstation/Playstation 2 support.

Pazureka, 2023-10-13:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

Provide support for Playstation/Playstation2 consoles for easy all in one solution.

### Describe the solution you would like to see

Because RP2040 has 30 free pins on board, i think it would be neat to add support for PSX/PS2 since controller needs 7 pins. With standard aracade stick/hitbox there would be plenty of pins left to do it. 
[image]

Someone already made controller emulator on arduino, so it's possible. [https://github.com/veroxzik/arduino-psx-controller](url)
### Describe alternatives you've considered

An alternative would be designing controller/PCB with place for ardunio board separately and adding diodes to seperate signals between RP2040 and Arduino.

### Can you submit a pull request?

No

### Are you planning on working on this feature yourself?

I cannot code, sorry.

TheTrainGoes, 2023-10-13:
Hey @Pazureka ,

While I agree it would be neat to have this support we mainly rely on people making these additions and sharing the code with us.  

As such I don’t believe we currently have anyone working on this but I am happy to leave the ticket open should anyone come along that can code it.

gromber, 2024-03-31:
Hi! hope they add macos dinput like 8bitdo did, because with that mode, you can use openps2loader to use ps3 emulator gamepad to play playstation 2 games. Now don´t work with normal dinput mode.

mikepparks, 2024-09-22:
PS2 input via OPL PADEMU will work with the upcoming 0.7.10 when using either PS3 or PS4 input modes.  PS4 mode will require using the "Remote/Emulation" input option in order to work in PADEMU.

Sometime in the future, we're looking at options to enable true PS1/2 input using the standard controller port.

TheTrainGoes, 2025-01-03:
We are closing all issues related to retro support and will be re-adding new tickets for the systems we are interested in looking at adding.

closed 2025-01-03 by TheTrainGoes