OpenStickCommunity/GP2040-CE #575: How to implement combine buttons

zl827154659, 2023-10-15:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/#/) to confirm this feature doesn't already exist?

Yes

### Which problem is this feature request solving?

I'm playing Street Fighter 6 and there is some new function buttons (like Drive Impact: HP+HK, Drive Parry: MP+MK, Throw: LP+LK).
Although I could set these combine buttons to LS or RS or BACK, but there still be some problems that in Training Room, LS and RS were occupied by Recording Actions and Play Actions so that I couldn't make these function be effect.

### Describe the solution you would like to see

Implement custom combine buttons pinout in configuration webpage, which could make some pinout have the functionality of combination buttons, pressing this pinout button is equal to pressing other multi pinout(or button) (like MP+MK). 

### Describe alternatives you've considered

Maybe there are some hardware solutions, but I think this would be very cool and useful to custom combine pinout in the future features.

### Can you submit a pull request?

No. Sorry that I'm very busy in my job.

### Are you planning on working on this feature yourself?

No. Except the above reason, I'm not a native English Speaker and my poor English limits the effective communication. But I'm learning RP development based on my Java, Python, C ability.

TheTrainGoes, 2023-10-15:
This is all possible with the new macro system.  You will be able to set an unused pin to trigger a dedicated macro that is mapped to the combos of buttons.

zl827154659, 2023-10-15:
> This is all possible with the new macro system. You will be able to set an unused pin to trigger a dedicated macro that is mapped to the combos of buttons.
Thanks for replying. Did macro system be released ? I cant find macro setting in my configuration webpage (version: 0.7.4) although I could see the doc does mention about it.
 https://gp2040-ce.info/#/macros

zl827154659, 2023-10-16:
It seems that the macro system fixed my most requirements, but I have to say there still are some little problems.
For example, Drive Rush in SF6 is triggered by pressing twice Foward/Right while the character parrying, which means I need to keep holding parrying (MK + MP) and press twice Right. Then the problem comes: How could I use macro system to implement "holding to keep pressing (not holding to repeat)". 
At first I tried to set up longer Input Line Duration and set 'Interruptible' on and switch the Trigger mode to 'Holding to repeat'  but of course the first pressing Right would interrupt parrying status and failed to Drive Rush.
Obviously that 'Interruptible' should be set off in my requirement, but I dont need the long Input Line Duration, I just need it to be holding to keep pressing multi buttons,  release to stop multi buttons so.
Above all, I think there should be a new trigger mode like 'holding to keep pressing' not to repeat. This seems not a macro func, maybe a new feature?

TheTrainGoes, 2023-10-16:
Can all of this be addressed by changing what does the record and replay in training?

jack2game, 2023-10-16:
> I need to keep holding parrying (MK + MP) and press twice Right. Then the problem comes: How could I use macro system to implement "holding to keep pressing (not holding to repeat)".

TBH this is a very game-specific input, thus not sure if it warrants a whole different feature with the current macro function which is already complex enough. Furthermore, Drive Rush can be achieved in different way: You can press 1) Right, and then 2) Right+Parry. It's quicker this way, saves drive gauge and also easier to implement.

zl827154659, 2023-10-16:
> Can all of this be addressed by changing what does the record and replay in training?

Recording and Replaying buttons conflict was fixed by current macro system. 'Holding to keep pressing' is a new question with it.

zl827154659, 2023-10-16:
> TBH this is a very game-specific input, thus not sure if it warrants a whole different feature with the current macro function which is already complex enough. Furthermore, Drive Rush can be [...]

lol, I totally understandand and thanks again for all ur contribution.
BTW, in some scenes, like the opponent was  using Hadouken continuously. Parry and drive rush is a simple way to break it but we may not remember to input the first Right :)
Now I use another way to fix the requirement by replace parry to other button. And using macro to implement Drive Impact(cause it needs just press once)
Anyway, I just offer an idea about the extra button inputs. Of course gp2040 is a universal project and should not be game-specific, I truly admire u guys.

closed 2023-10-16 by zl827154659