OpenStickCommunity/GP2040-CE #597: down and left/right inputs get randomly stuck

skreamz, 2023-10-24:
### Prerequisites

Please check the following before posting an issue / bug report.

- [check ] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [check ] I have checked the documentation [HERE](https://gp2040-ce.info/#/) and found no answer
- [check ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [check ] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: [v0.7.5]
* Name of device: rpi pico
* Link to where you purchased the device: amazon
* Is this a custom build?: no
* Operating System: windows 11
* Browser: ? really? chrome

### Expected Behavior

Please describe the behaviour you are expecting.

### Current Behavior

What is the current behaviour?

I'm playing a lot of fighting games, mainly sf6, holding down, and back which is left or right, Sometimes the inputs get stuck, and I have to unplug and plug back in to clear it, it happens randomly, sometimes not at all, sometimes a few times in an hour. settings are not touched after installing the firmware too.

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

too random to reproduce constantly

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

TheTrainGoes, 2023-10-24:
Hi @skreamz ,

this is not something we have had reported.  Are you able to post your backup file from web-config?  If you are on our Discord you can also shoot it over there.

skreamz, 2023-10-24:
is this the file you are on about?
[gp2040ce_backup_20231024165939994.zip](https://github.com/OpenStickCommunity/GP2040-CE/files/13121778/gp2040ce_backup_20231024165939994.zip)

TheTrainGoes, 2023-10-24:
Yes that is the file I was on about.  

Was this a fresh install of v0.7.5 or did you upgrade from another release?  

I would recommend that you nuke the Pico and do a fresh install of v0.7.5.

I have tested your backup file on my Pico running a fresh copy of v0.7.5 (post nuke) and cannot replicate the issue.

TheTrainGoes, 2023-11-23:
Have not gotten a reply back on this so I am closing the request.

closed 2023-11-23 by TheTrainGoes