OpenStickCommunity/GP2040-CE #609: How to get past this screen?

d0n13, 2023-11-03:
Hi,

I'd ask on Discord but server link is expired...

Have setup the GP2040 to be a PS4 stick with PS4 Mode enabled. 

However, I notice the code has the following in the PS4 descriptors but would it make sense to use the DS4 vendor and product ids like 0x054c and 0x09cc ??

```
// Razer Panthera
#define PS4_VENDOR_ID		0x1532
#define PS4_PRODUCT_ID		0x0401
```

I'm trying to get past the connect controller screen as shown below but it's not working. The PS button does work once in the menus etc. Is there a way for the PS4 to allow this controller to be used on its own without having to have a real DS4 to get past this screen?

[image]

Thanks and sorry for having to post this here...

D

d0n13, 2023-11-03:
Strongly, the PS allows the GP2040 to be used in safe mode and accepts the PS key there to get past the connect controller screen. I can navigate the menus of same mode with no issues.

TheTrainGoes, 2023-11-03:
Hi @d0n13 ,

There should be no issues with the server link - https://discord.com/invite/k2pxhke7q8

This is a rather long standing issue that has been around since the PS3.  Many 3rd party controllers simply do not work on this screen for unknown reasons.  

It is recommended to use a genuine DS4 for doing this.

TheTrainGoes, 2023-11-23:
Closing this as there is nothing we can do to fix it.

closed 2023-11-23 by TheTrainGoes