OpenStickCommunity/GP2040-CE #618: PS4 mode display + LED bug

ez2torta, 2023-11-10:
### Prerequisites

Please check the following before posting an issue / bug report.

- [X] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [ ] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [ ] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [ ] I am reporting the issue to the correct repository

### Context

* Firmware Version: 0.7.5
* Name of device: SKY2040
* Link to where you purchased the device: AliExpress
* Is this a custom build?: kind of
* Operating System: ps4
* Browser: ?

### Expected Behavior

Please describe the behaviour you are expecting.

I'm using the controller in ps4 mode with rgb buttons and controller display. I noticed today that randomly the rgb and display would freeze

### Current Behavior

What is the current behaviour?
Sometimes I keep pressing buttons and they do register in game but don't register in the display neither rgb

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Play in ps4 mode
2. Go into your game and press some random buttons in succession
3. Randomly the display and rgb would lag/stutter/freeze

### Screenshots & Files

Please include any relevant screenshots or files here.  If you are having issues with a board please include images of the front and back of the board.

https://youtube.com/shorts/aDJVyYN8enU?si=sOfO2ra_CBMRlui1

I thought it would be hard to replicate but actually took only a minute. Any suggestions?

TheTrainGoes, 2023-11-10:
Hi @ez2torta,

This is expected behavior when you are using the device in PS4 mode with the necessary files uploaded to use without an 8min timeout.   

The regular authentication that GP2040-CE needs to do to stay connected to the PS4 is very taxing on Core1 which leads to the freezing of the screen and or LeDs.

closed 2023-11-10 by TheTrainGoes

ez2torta, 2023-11-10:
woah, is this documented then?

should i use it with a brook ps5 wingman converter to avoid any freezes?

TheTrainGoes, 2023-11-10:
@ez2torta - to be clear - this does not affect Core0 performance at all.  It will not impact your inputs.   

You are welcome to use a Wingman if the LEDs are important to you.  

We also have a new way to authenticate which is by USB passthrough however the device you have will not support it.

ez2torta, 2023-11-10:
Yeah, This is my first time using leds/display so i was worried it impacted gameplay
i've been messing with gp2040 since 2022 and it has been a long way. I really appreciate you answering all of this :)