OpenStickCommunity/GP2040-CE #622: Issue when Saving Macros

ZoddoDK, 2023-11-15:
I am running version Pico Fighting Board (GP2040-CE_0.7.5_PicoFightingBoard.uf2)

When i want to save any Settings on the Macro Page via the Save button nothing is happening and the settings are apparently not saved.
If I refresh the page after, everything is back to the Default Settings.

ZoddoDK, 2023-12-04:
I double checked again, this time I assigned a Macro to Pin 21, which I double checked to not be assigned anywhere in the Pin mapping, and I still have he same result.
When I bind Pin 21 in Pin Mappings to L3 it works just fine.

TheTrainGoes, 2023-12-04:
Please take a screen shot of your Macro setup prior to attempting to press the save button.  

Also please let us know what device you are trying to do this on and include a copy of your backup file.   

If you are on the Discord it may be easier to file a ticket there.

ZoddoDK, 2023-12-05:
[image]
[image]
Now i press Save. No Message or anything appears.
After I refresh the Page looks like this:
[image]

ZoddoDK, 2023-12-05:
[gp2040ce_backup_20231205042415855.zip](https://github.com/OpenStickCommunity/GP2040-CE/files/13555100/gp2040ce_backup_20231205042415855.zip)

ZoddoDK, 2023-12-05:
The device is a Pico Fighting Board

TheTrainGoes, 2023-12-05:
In that screen shot you have not activated the macro.  Please try activating the macro

ZoddoDK, 2023-12-05:
[image]
That doesn't seem make any difference, settings still will not get saved when I press the save Button.

TheTrainGoes, 2023-12-05:
I loaded your backup and it appears there was some sort of saved pin issue on pin 20 and 21.  

I was able to do the following with your backup to resolve:

1 - Nuke
2 - Load PFB UF2
3 - Load your backup
4 - Change the extra button pin from 20 to -1, then disable and save
5 - Setup the macro on pin 21 and save
6 - Turn back on the extra button pin on 20 and save

I was also able to do this on a fresh install of the firmware directly after a nuke.

ZoddoDK, 2023-12-05:
Thanks for all the effort. I will give that a try.

ZoddoDK, 2023-12-05:
I did everything as described but still have the same issue. The only thing I wasn't sure about was the "Nuke" part. What I did was Boot in "USB (BOOTSEL)" and then I reinstalled "GP2040-CE_0.7.5_PicoFightingBoard.uf2". Does that count as a "Nuke"?
Afterward I did 3-4-5-6 with the same result in the end.

ZoddoDK, 2023-12-06:
OK I will try that. Thanks.

ZoddoDK, 2023-12-06:
Thanks, a lot. That did the trick.

closed 2023-12-06 by ZoddoDK