OpenStickCommunity/GP2040-CE #624: Expand Dual Directional Input to triple input and allow both analogs and the D-pad simultaneously

TheOpponent, 2023-11-15:
### Have you checked the GP2040-CE documentation [HERE](https://gp2040-ce.info/) to confirm this feature doesn't already exist?

**Yes**

### Which problem is this feature request solving?

I'm building a controller that exposes all inputs of a standard gamepad, and the analog sticks as digital joysticks. In addition to the Dual Directional Input add-on accepting digital inputs as analog controls at maximum range, I want to also allow discrete D-pad input as well. The primary use case is for games that use two analog sticks and the D-pad for ancillary functions, such as twin-stick shooter games. I'm using a standard Raspberry Pi Pico, and as I am not using any other add-on hardware, I was hoping to use four more GPIO pins for D-pad input.

### Describe the solution you would like to see

Expand the Dual Directional Input add-on to allow left analog, right analog, and the D-pad all at once, or expand the Extra Button add-on to accept more inputs.

### Describe alternatives you've considered

I attempted to leverage the Joystick Selection Slider to momentarily change the directional inputs from left analog to D-pad when D-pad buttons are pressed, but found that there is noticeable lag when changing modes this way.

### Can you submit a pull request?

**Yes**

### Are you planning on working on this feature yourself?

**No**, as I do not have enough experience in embedded hardware development to expand the current add-ons for this purpose.

InfraredAces, 2023-11-15:
It sounds like you are better served using the [Tilt Input ](https://gp2040-ce.info/add-ons/tilt-input) add-on as that allows you to have digital inputs for left and right analog stick.

TheOpponent, 2023-11-15:
> It sounds like you are better served using the [Tilt Input ](https://gp2040-ce.info/add-ons/tilt-input) add-on as that allows you to have digital inputs for left and right analog stick.

I don't currently possess any analog joysticks to use with this add-on. I want to send digital input, not analog.

InfraredAces, 2023-11-16:
Please reread my statement and the [Tilt Input ](https://gp2040-ce.info/add-ons/tilt-input) add-on documentation for clarity on how to achieve what you have stated as your goal. The use of **digital inputs** on Left analog stick, Directional Pad, and Right analog stick simultaneously.

1. The add-on allows you to send **digital input** to the LS and RS simultaneously. 
2. The D-Pad is **only deactivated if you use the hotkeys to change the DPad mode to anything other than DPad Digital.
> using the hotkeys DPad Left Analog and DPad Right Analog deactivates the DPad and using the DPad Digital will reactivate the DPad once more.

TheOpponent, 2023-11-16:
Thank you for the clarification. I understand now, I left the Tilt 1 and Tilt 2 pins mapped to -1 because I wasn't planning on using them in my controller, but unless at least one of them is set to a valid pin, the rest of the Tilt pins don't function at all. I misunderstood in the documentation that at least one of Tilt 1 and Tilt 2 has to be set even if they won't be used in order to use eight discrete Tilt Left and Tilt Right pins. Perhaps the documentation in the add-ons page can be updated to make this requirement more explicit?

TheTrainGoes, 2023-12-10:
Hi @TheOpponent ,

Please feel free to submit a PR with the changes you think you make this information more clear.

closed 2023-12-10 by TheTrainGoes