OpenStickCommunity/GP2040-CE #642: Can not open Web Configurator by 192.168.7.1

cjsliuj, 2023-11-27:
### Prerequisites

Please check the following before posting an issue / bug report.

- [Yes] I am running the latest version of the firmware found [HERE](https://github.com/OpenStickCommunity/GP2040-CE/releases)
- [Yes] I have checked the documentation [HERE](https://gp2040-ce.info/) and found no answer
- [Yes] I checked to make sure that this issue has not already been filed [HERE](https://github.com/OpenStickCommunity/GP2040-CE/issues)
- [Yes] I am reporting the issue to the correct repository

### Context

Please provide all relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.  

* Firmware Version: 0.7.5
* Name of device: Raspberry Pi Pico
* Link to where you purchased the device: https://ic-item.jd.com/10073555936788.html
* Is this a custom build?: NO
* Operating System: Windows11
* Browser: chrome

### Expected Behavior

open the web configure
### Current Behavior
not open ,with a error message "ERR_EMPTY_RESPONSE"

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. just connect GP17 和 GND pin
2. plugin in to pc
3. open 192.169.7.1 in chrome

### Screenshots & Files
ipconfig
[image]

wiring
[image]

brower
[image]

DaoraDio, 2023-11-29:
Try these steps:
* Your button could be connected wrong. Always connect diagonal pins to ensure that you are not connecting GND to GND or GP17 to GP17. Or just remove the button altogether and connect GP17 to GND directly with a female to female jumper wire
* Ensure that your firewall is not blocking anything, turn of temporarily to check
* Use a different browser

cjsliuj, 2023-11-30:
fixed thank you